Jaeyoung-Lim/mavros_controllers #2: [Feature] Trajectory Generation

Jaeyoung-Lim, 2018-07-15:
[mav_trajectory_generation](https://github.com/ethz-asl/mav_trajectory_generation) looks like a good framework for the 'generation' of trajectories.

A demo example and necessary interfaces should be created
[ ] Try trajectory generation
[ ] Create interface for trajectory representation
[ ] Verify this is working together

marcusabate, 2019-03-05:
@Jaeyoung-Lim I've been working on a project utilizing [waypoint_navigator](https://github.com/ethz-asl/waypoint_navigator) as the framework for generating trajectories. You may remember michaelkapteyn did this as well.

If you'd like to take a look at [my fork](https://github.com/marcusabate/mavros_controllers) you'll see the necessary changes to implement this. I can create a pr if you'd like.

Jaeyoung-Lim, 2019-03-05:
@marcusabate That is a nice implementation! A PR will be appreciated :smile:

closed 2019-03-19 by Jaeyoung-Lim