Jaeyoung-Lim/mavros_controllers #34: How to use this code

weiqil, 2018-10-30:
Hi,
Thank you for your sharing of this work. It does help me to understand the papers much. But when I try to run this code on my computer, there comes some errors.
I type this command:
roslaunch geometric_controller trajectory_track_circle.launch 
 Part of the output is:
UAV autoquad icarous matrixpilot paparazzi slugs standard uAvionix ualberta
[ INFO] [1540910990.048991065]: MAVROS started. MY ID 1.240, TARGET ID 1.1
0x2002e10 void QWindowPrivate::setTopLevelScreen(QScreen*, bool) ( QScreen(0x192de90) ): Attempt to set a screen on a child window.
0x1fff110 void QWindowPrivate::setTopLevelScreen(QScreen*, bool) ( QScreen(0x192de90) ): Attempt to set a screen on a child window.
0x2029880 void QWindowPrivate::setTopLevelScreen(QScreen*, bool) ( QScreen(0x192de90) ): Attempt to set a screen on a child window.
0x2028be0 void QWindowPrivate::setTopLevelScreen(QScreen*, bool) ( QScreen(0x192de90) ): Attempt to set a screen on a child window.
[ERROR] [1540910994.865435761]: MODE: Unsupported FCU
[ERROR] [1540910999.875788522]: MODE: Unsupported FCU
[ERROR] [1540911004.886073662]: MODE: Unsupported FCU

Is there any thing wrong for my operation?
Thanks a lot!

Jaeyoung-Lim, 2018-10-31:
@weiqil Thanks for the interest,

I guess the repo is currently badly documented. I will include a brief tutorial on how to setup the environment. The current simulation is based on the [PX4 SITL](https://dev.px4.io/en/simulation/#sitl-simulation-environment)

It seems like your SITL is not running properly, as in this [document](http://dev.px4.io/en/simulation/ros_interface.html)

You need to check if the following command is working properly. 
```
cd <Firmware>
make posix_sitl_default gazebo
```
Once this is working, you can run the ROS node in this repo

Jaeyoung-Lim, 2018-10-31:
@weiqil There seems to be something wrong with your firmware installation.

Try 
```
cd <Firmware_clone>
make posix_sitl_default gazebo
source ~/catkin_ws/devel/setup.bash    // (optional)
source Tools/setup_gazebo.bash $(pwd) $(pwd)/build/posix_sitl_default
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)/Tools/sitl_gazebo
roslaunch px4 posix_sitl.launch
```

closed 2018-11-12 by Jaeyoung-Lim

Jaeyoung-Lim, 2018-11-21:
@weiqil Could you confirm that the comment solved your problem?

weiqil, 2018-11-23:
@Jaeyoung-Lim  
Hi, 
 Sorry I'm late reply.  Indeed, I can not run the code by  the launch file  all the time.　
$ roslaunch geometric_controller trajectory_track_circle.launch
  parts of the output are:

process[sitl-5]: started with pid [5422]
INFO  [Unknown] PX4 daemon already running for instance 0 (Resource temporarily unavailable)
process[gazebo-6]: started with pid [5423]
process[gazebo_gui-7]: started with pid [5428]
process[vehicle_spawn_Asus_4746_4639041922856050695-8]: started with pid [5433]
process[rviz-9]: started with pid [5434]
================================================================================REQUIRED process [sitl-5] has died!
process has died [pid 5422, exit code 255, cmd /home/weiqi_liu/src/Firmware/build/posix_sitl_default/bin/px4 /home/weiqi_liu/src/Firmware/ROMFS/px4fmu_common -s etc/init.d-posix/rcS __name:=sitl __log:=/home/weiqi_liu/.ros/log/20f914f6-eecf-11e8-9d80-1c872c47930c/sitl-5.log].
log file: /home/weiqi_liu/.ros/log/20f914f6-eecf-11e8-9d80-1c872c47930c/sitl-5*.log
Initiating shutdown!

But if I run the code step by step, I get the desired result that the vehicle takes off and then follows a circular trajectory. The main steps are:

- **run the gazebo SITL**

 make posix_sitl_default gazebo

- **run the mavros**

 roslaunch mavros px4.launch fcu_url:="udp://:14540@127.0.0.1:14557"

- **run the controller**

rosrun geometric_controller geometric_controller 

- **run the trajectory publisher**

rosrun trajectory_publisher trajectory_publisher 

Sorry  for my late reply and thanks for your sharing and help again. 

  weiqi liu

stijojoseph, 2020-10-07:
when running make posix_sitl_default gazebo

Makefile:493: *** "Make target posix_sitl_default not found. It either does not exist or posix_sitl_default cannot be the first argument. Use 'make help|list_config_targets' to get a list of all possible [configuration] targets.".  Stop

commit e7c5253a (2018-11-12): Update running instructions

This fixes #34 #36