Jaeyoung-Lim/mavros_controllers #41: New Firmware Problems

marcusabate, 2019-02-19:
Hello,

I am using your implementation of the geometric controller for the px4 to control a drone in simulation and hopefully in real experiments. However, I'm running into a few issues related to px4 Firmware, Mavros etc.

Using the most up-to-date version of the px4 Firmware and of mavros causes several issues with this package. To begin, using the gcs_url default argument you specify in every launch file gives me an error to the effect of ``Local frame 18 not supported. Unable to publish pose and velocity.``If I simply comment out that argument I'm fine here.

The next problem is that running the mavros node like that and attempting to switch to offboard gives me the following error:
```
[FATAL] [1550545478.242996294, 5.016000000]: Recieved thrust, but ignore_thrust is true: the most likely cause of this is a failure to specify the thrust_scaling parameters on px4/apm_config.yaml. Actuation will be ignored.
```
The drone fails to takeoff. I can arm from px4 shell but mode switch to offboard fails.
I believe this is because newer versions of mavros and px4 Firmware require you to load a px4 config yaml file in the mavros directory. If I add that to your launch files then the drone is able to arm, switch to offboard and takeoff as intended. However, even then the controller doesn't work as you'd hope and drives the drone crazy until it crashes and flips over. I can't get it to track any trajectory.

If I try to get it to simply takeoff and hover at [0, 0, 2] position, it'll still veer off to a side and crash. I suspect it might be a coordinate frame mismatch?

I cannot revert to your fork of the firmware because my application requires newer code. Any thoughts on what might be causing these issues? All three of them stump me because I don't find many references to the Local frame 18 thing or the requirement to use the px4 config file, so my fixes seem unneeded to me. Any help would be appreciated!

marcusabate, 2019-02-19:
@Jaeyoung-Lim I am building from source.

Jaeyoung-Lim, 2019-02-20:
@marcusabate Okay cool. I will be able to look into this during the weekend. Contributions would also be welcome!

Jaeyoung-Lim, 2019-02-20:
@marcusabate Okay cool. I will be able to look into this during the weekend.

Jaeyoung-Lim, 2019-03-01:
@marcusabate I have made a fix for this. Could you confirm that this is working for you?

marcusabate, 2019-03-05:
@Jaeyoung-Lim I can confirm that this is working with gazebo_states enabled.

However, if I disable gazebo_states and rely on /mavros/local_position/..., the drone behaves erratically. This is because /mavros/local_position/velocity is not publishing any data on my machine. I'm not sure if this is something that mavros master has updated or if this is just me. I'll be taking a look later this week.

Could you give me a link to the mavros commit where they changed the coordinate frame? I hadn't been able to find it as quickly as you did.

Jaeyoung-Lim, 2019-03-05:
@marcusabate I realized that too.

I tried to dig in but it seems to be somewhere related to this PR: https://github.com/mavlink/mavros/pull/1121 but doesn't seem obvious why this would change the coordinate frame. 

The wierd behavior with local estimates were also what I am observing. I would like to look into this but

PR #47 (2019-03-01): Fix regression with current mavros master
There has been a regression on this repo with the current mavros master.

This PR includes the following fixes
- Fix on a coordinate system mismatch on the mavros side
- Reading the px4_config file from mavros by default

This fixes #41 

@marcusabate Hope this helps
PR #56 (2019-03-06): Fix regression on topic name change from mavros
This fixes  #41 where the controller was not behaving properly with local position estimates as reported by @marcusabate

The regression was coming from a recent PR https://github.com/mavlink/mavros/pull/1160 in mavros, which the topic name changed and there were no longer a velocity topic being subscribed by the controller. 

The changes that was made is as the following.
- Changed subscriber topic name following the mavros PR
- Lowered the velocity gains for reference tracking

This also introduces an issue where the mavros topic names now diff [... cut]
closed 2019-03-06 by Jaeyoung-Lim