Jaeyoung-Lim/mavros_controllers #58: Reason for the error

shivamgarg1901, 2019-03-09:
Hi Jaeyoung-Lim!
I tried to run the code as explained in the repository.The packages builds up perfectly but when I  use the roslaunch command then it shows one error which is as under-
[mavros-4] process has died [pid 10798, exit code -11, cmd /home/shivam/catkin_ws1/devel/lib/mavros/mavros_node __name:=mavros __log:=/home/shivam/.ros/log/9ef07a40-423d-11e9-b232-707781c0679d/mavros-4.log].
log file: /home/shivam/.ros/log/9ef07a40-423d-11e9-b232-707781c0679d/mavros-4*.log
As a virtue of which my drone remains stationary in gazebo.
Please help me with it.
By the way you did great work!
Thanks in advance.

Jaeyoung-Lim, 2019-03-09:
@shivamgarg1901 Thank you for the kind words

Could you clarify on which launch file you used?

shivamgarg1901, 2019-03-09:
> @shivamgarg1901 Thank you for the kind words
> 
> Could you clarify on which launch file you used and also post the full log of your command?

shivamgarg1901, 2019-03-09:
roslaunch geometric_controller trajectory_track_circle.launch
After executing above command I got the error on the terminal. 
[image]
Also after executing the command roslaunch mavros px4.launch.I got the error on the terminal similar to the above
[image]
Please! help me

Jaeyoung-Lim, 2019-03-09:
Are you trying to connect to the fcu on USB?

shivamgarg1901, 2019-03-09:
No I just want to do it on gazebo.

Jaeyoung-Lim, 2019-03-09:
@shivamgarg1901 Your log says your launchfile is configured to connect to USB

shivamgarg1901, 2019-03-09:
I followed this link https://dev.px4.io/en/ros/mavros_installation.html
Is there any way I can remove the existing mavros and reinstall the new one which will fulfill my requirement?

Jaeyoung-Lim, 2019-03-09:
@shivamgarg1901 My question was more on if you installed it from source or the binary using `apt` as the binary package is outdated.

shivamgarg1901, 2019-03-09:
Yes I did the binary installation.

Jaeyoung-Lim, 2019-03-09:
@shivamgarg1901 You need to install it from source. Sorry for not being clear on the documentation. This was due to a recent regression on the mavros side.

shivamgarg1901, 2019-03-09:
It's completely fine can you please give me the sequence of commands to do source installation? Also How could I get rid of that already installed binary mavros?

Jaeyoung-Lim, 2019-03-09:
@shivamgarg1901 It is in the documentation you were referring to. To remove, try

```
apt remove ros-kinetic-mavros*
```

shivamgarg1901, 2019-03-11:
[image]
I did the source installation of the mavros so the previous errors are now removed but now the terminal shows this error.How could I handle this error.Please help.

Jaeyoung-Lim, 2019-03-11:
@shivamgarg1901 Somehow the geometric controller launchfile is not reading the config file. Are you sure you are using the latest master of this repo?

shivamgarg1901, 2019-03-12:
@Jaeyoung-Lim It is working fine now.Thanks for your help.You did fantastic job :)

Jaeyoung-Lim, 2019-03-12:
@shivamgarg1901 Thats great! Glad that it works.

closed 2019-03-12 by Jaeyoung-Lim