Jaeyoung-Lim/mavros_controllers #61: Compilation Problem

altshuler, 2019-03-17:
Hello, 
I'm trying to compile the mavros_controllers and get compilation error:

Errors     << geometric_controller:check /home/evgeny/catkin_ws/logs/geometric_controller/build.check.017.log                
CMake Error at /home/evgeny/catkin_ws/devel/share/catkin_simple/cmake/catkin_simple-extras.cmake:38 (find_package):
  By not providing "Findeigen_catkin.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "eigen_catkin", but CMake did not find one.

  Could not find a package configuration file provided by "eigen_catkin" with
  any of the following names:

    eigen_catkinConfig.cmake
    eigen_catkin-config.cmake

  Add the installation prefix of "eigen_catkin" to CMAKE_PREFIX_PATH or set
  "eigen_catkin_DIR" to a directory containing one of the above files.  If
  "eigen_catkin" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:7 (catkin_simple)

make: *** [cmake_check_build_system] Error 1

What you think is a problem ?

Best regards
Evgeny

Jaeyoung-Lim, 2019-03-17:
This is a recently added dependency. Sorry for not updating the readme.
Try installing it through the `dependeincies.rosinstall` in the root of the repo
 ```
cd ~/catkin_ws
wstool merge -t src mavros_controllers/dependencies.rosinstall
wstool update -t src -j4
rosdep install --from-paths src --ignore-src -y --rosdistro kinetic
```
or you can simply clone it to the src directory as the following
```
git clone https://github.com/ethz-asl/eigen_catkin.git
```

altshuler, 2019-03-18:
Thank you very much.
This commands will update only mavros_controllers package or
all packages in workspace ? I have a packages that I don't want to update.

BR
Evgeny

Jaeyoung-Lim, 2019-03-18:
@altshuler If you are worried about dependencies, simply clone `eigen_catkin` into your workspace. That should be enough

altshuler, 2019-03-19:
Hello,
when I do   roslaunch geometric_controller trajectory_track_circle.launch    command.
I get this message: 
ERROR: cannot launch node of type [trajectory_publisher/trajectory_publisher]: can't locate node [trajectory_publisher] in package [trajectory_publisher]

The drone just takeoff and hovering in one place.

Jaeyoung-Lim, 2019-03-19:
Did you build the `trajectory publisher`?

altshuler, 2019-03-19:
Actually, no...
I didn't get that needed to build it separately.
Any other packages should be built separately ?

Jaeyoung-Lim, 2019-03-19:
As written in the instructions, if you build `mavros_controllers` it will build `trajectory_publisher` and `geometric_controller`

altshuler, 2019-03-19:
Ok, thanks.
Will test it today.

Jaeyoung-Lim, 2019-03-24:
@altshuler Any updates?

altshuler, 2019-03-25:
I'm compiled the packages, Running simulations ,
trying to understand the code.
I have 2 questions.
Ho I can disable running Gazebo every simulation ?
What launch file runs real flight (not simulation) ?

Jaeyoung-Lim, 2019-03-25:
@altshuler So does that mean the problem is solved?

On a real platform, you need to disable `sim_enable_` parameter and not launch `px4_sitl`.

Currently there is no launch file specifically for a real system. Do you need one?

altshuler, 2019-03-25:
Yes, the problem is solved.
I will try to fly on real drone.

Tanks

Jaeyoung-Lim, 2019-04-08:
@altshuler Have you tried flying?

altshuler, 2019-04-08:
Actually, yes. But the vehicle is not arming.
I disabled sim_enable_ parameter  and px4_sitl .
And I tried it before last updates that you added to the project.
Unfortunately I have some HW problems on my Intel Aero, so I will try it 
on next week. 

Can you explain in couple of words what dob_controller do ?

Jaeyoung-Lim, 2019-04-08:
Cool. Hope it works :smile: 

A dob_controller is a implementation of a disturbance observer based control. It tries to estimate the disturbance and compensate it. This is at a very experimental stage. It works in SITL, but flies a bit aggressively which I am not sure how it would behave on a real drone.

Jaeyoung-Lim, 2019-04-10:
Closing as the issue is resolved

closed 2019-04-10 by Jaeyoung-Lim