Jaeyoung-Lim/mavros_controllers #63: Determining Constant Values

ScoobyCodes, 2019-03-27:
HI! I was determining how are you determining value of constants like Kpos, Kvel and att_ctrl_tau.

Jaeyoung-Lim, 2019-03-27:
@ScoobyCodes It is a tuning parameter. It decides how aggressive you want to control your vehicle.

It can depend on various things such as how noisy your estimate is, how agile your vehicle is .etc

Jaeyoung-Lim, 2019-03-30:
@SourabhGupta12 Does this answer your question?

SourabhGupta12, 2019-03-30:
Hi! I was actually looking for some mathematical Model to determine these constants. Have you used gradient descent and optimized the possible error in your control model ?

Jaeyoung-Lim, 2019-03-30:
@SourabhGupta12 
I did not use any optimization as these are simply parameters that tune the behavior of the controller
For example, `att_ctrl_tau` is a time constant given that we are sending angular rate commands so this is more to tune on how aggressive your attitude controller is. 

If you want to do gradient descent, you first need to define a cost to define the optimality of the gain. 

In case you want to define your own metric to define what is a more desirable behavior, I believe there are plenty of literature on optimal control that looks into this problem

closed 2019-04-10 by Jaeyoung-Lim