Jaeyoung-Lim/mavros_controllers #69: trajectory_msgs/MultiDOFJointTrajectoryPoint.h : No such file or directory

BhavyanshM, 2019-03-30:
Hi @Jaeyoung-Lim 

On launching a ```catkin build```, an error related to a header file in trajectory_msgs pops up. I checked the common_msgs/trajectory_msgs, there is not ```MultiDOFJointTrajectoryPoint.h``` in there. Even on the official repository for trajectory_msgs ([https://github.com/ros/common_msgs/tree/jade-devel/trajectory_msgs](https://github.com/ros/common_msgs/tree/jade-devel/trajectory_msgs)) I could not find that header file. Do you think this file belongs to a different package?

Jaeyoung-Lim, 2019-03-30:
You are right It is not a standard package.

you can get it here: https://github.com/ethz-asl/mav_comm.git

BhavyanshM, 2019-03-30:
> You are right It is not a standard mesaage
> 
> you can get it here: https://github.com/ethz-asl/mav_comm.git

So it turns out that I already have ```mav_comm``` in my catkin_ws. 
```
~/catkin_ws/src/mav_comm $ git remote show origin 
* remote origin
  Fetch URL: https://github.com/ethz-asl/mav_comm.git
  Push  URL: https://github.com/ethz-asl/mav_comm.git
  HEAD branch: master
  Remote branches:
    feature/baro_msg_by_burri                       tracked
    feature/default_speed_topics                    tracked
    feature/manipulator_msgs                        tracked
    feature/map_api_service_for_planner             tracked
    feature/roll_pitch_yawrate_msg_with_feedforward tracked
    feature/rotation_vectors                        tracked
    feature/rpy_helper_functions                    tracked
    feature/timestamp                               tracked
    feature/trajectoryArrayMsg                      tracked
    feature/waypoint_navigator_integration          tracked
    master                                          tracked
    mav_standardization                             tracked
    release_changes                                 tracked
    set_flight_mode                                 tracked
    v3.2.0                                          tracked
  Local branch configured for 'git pull':
    master merges with remote master
  Local ref configured for 'git push':
    master pushes to master (up to date)

```

However, I don't see any ```trajectory_msgs``` inside ```mav_comm``` either. Even with the ```mav_comm``` up-to-date with the master on the repository you posted, it throws the same error. Any suggestions?

Jaeyoung-Lim, 2019-03-30:
Oh, sorry about that I must have confused this. Which ros distribution are you using?

Putting the flight controller into offboard mode shouldn't take too long

BhavyanshM, 2019-03-30:
> Oh, sorry about that I must have confused this. Which ros distribution are you using?
> 
> Putting the flight controller into offboard mode shouldn't take too long

Oh no problem at all. So, I am currently using ROS Kinetic on Raspbian Stretch

```pi@AirStone:~/catkin_ws/src/mav_comm $ rosversion -d
kinetic
pi@AirStone:~/catkin_ws/src/mav_comm $ lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 9.6 (stretch)
Release:	9.6
Codename:	stretch```

I also do have ```common_msgs``` in my catkin_ws, however the ```trajectory_msgs``` inside common_msgs does not seem to have the MultiDOFJointTrajectoryPoint.h either. 

```
~/catkin_ws/src/common_msgs/trajectory_msgs $ ls
CHANGELOG.rst  CMakeLists.txt  msg  package.xml  rules
```

Jaeyoung-Lim, 2019-03-30:
Okay, that is interesting.

Is there a reason you are using raspbian stretch? ROS support in raspbian is in general poor. Which raspberry pi are you running ROS on?

BhavyanshM, 2019-03-30:
It's just that we started building the system on Raspbian Stretch and due to shortage of time were too scared to start fresh on a different distribution due to a large number of packages. We are currently running almost a hundred packages inside catkin_ws just fine. It's just that the header file for this message type seems to be missing.

BhavyanshM, 2019-03-30:
> Okay, that is interesting.
> 
> Is there a reason you are using raspbian stretch? ROS support in raspbian is in general poor. Which raspberry pi are you running ROS on?
> 
> One workaround would be just trying to build the trajectory_publisher with
> 
> ```
> catkin build trajectory_publisher
> ```
> which don't need the message.
> You can send `setpoint_raw` position target messages using the launchfile `trajectory_setpointraw.launch` which will also enable you using offboard mode.
> 
> Are you also planning to fly with this?

Yup, we do have a complete quadcopter with Pixhawk and Raspberry Pi ready to fly. It has been tested in several other modes. Just the Offboard mode is now needed. The setpoints will be generated by our Path Planning algorithm. The quadcopter will be flown indoors since it is supposed to map buildings using a 2D LIDAR-based SLAM algorithm. We successfully have Cartographer working as our SLAM algorithm.

Jaeyoung-Lim, 2019-03-30:
Can you try building just the trajectory publisher? May also comment out the geometric controller inside the `trajectory_setpointraw.launch` So you can simply test sending the setpoints to the flight controller.

BhavyanshM, 2019-03-30:
So yes, ```trajectory_publisher``` was able to build just fine. Okay, I will go ahead and try out the ```trajectory_setpointraw.launch``` without ```geometric_controller```. However, to send the setpoints actually from the Path Planning algorithm, do you think we would need the geometric_controller though?

Jaeyoung-Lim, 2019-03-30:
No, it depends on what setpoint you want to send.

the trajectory publisher will send a position and velocity reference which PX4 will try to track with a feedforward position control. This is the launch file running SITL: https://www.youtube.com/watch?v=6dLV01W0uEc

The geometric controller publishes body rate setpoints and run the attitude controller on the ROS side.

Jaeyoung-Lim, 2019-03-30:
Also, the launchfile will launch sitl, so you need to get rid of that part too.

BhavyanshM, 2019-03-30:
Hm, I see. Seems to work great on SITL definitely from the video. So if I understand it correctly, trajectory_publisher can be modified to use the position setpoints from any path-planning algorithm as well. In that case, let me gather the exact steps to launch launch the Offboard mode setup with trajectory_publisher right here. So that I can have your opinion on it as well. I'll go ahead and disable the geometric_controller and the SITL from the ```trajectory_setpointraw.launch``` and create a quick list of steps needed to launch Offboard mode. Just a min.

Jaeyoung-Lim, 2019-03-30:
Yes, currently the intention of the trajectory publisher was to use it to give example trajectories for the geometric controller. This package was made to have more aggressive control on the drone in offboard with the geometric controller.

However you can modify the trajectory pulblisher to publish whatever setpoints you want to make it follow if you add the right interface. Also contributions are welcome in case you add more functionalities :smile:

BhavyanshM, 2019-03-30:
> Yes, currently the intention of the trajectory publisher was to use it to give example trajectories for the geometric controller. This package was made to have more aggressive control on the drone [...]

Oh surely, its really impressive and fortunate that you have such a framework already in place. I would love to contribute with more sophisticated controllers as well as the proper launch files and configs needed to run your nodes on an actual Pixhawk based quadrotor.

BhavyanshM, 2019-03-31:
So @Jaeyoung-Lim I tried out the trajectory_setpointraw.launch after commenting out the geometry_controller node and the Posix SITL. However, the trajectory_publisher dies out with an error. Here is the complete log:

```
pi@AirStone:~/catkin_ws/src/mavros_controllers/trajectory_publisher/launch $ roslaunch trajectory_publisher trajectory_planner.launch 
... logging to /home/pi/.ros/log/64ba60aa-5347-11e9-b94a-b827eb79fcfb/roslaunch-AirStone-4335.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.0.27:40965/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/benewake_tfmini/field_of_view: 0.0
 * /mavros/distance_sensor/benewake_tfmini/frame_id: benewake_tfmini
 * /mavros/distance_sensor/benewake_tfmini/id: 0
 * /mavros/distance_sensor/benewake_tfmini/orientation: PITCH_270
 * /mavros/distance_sensor/benewake_tfmini/send_tf: True
 * /mavros/distance_sensor/benewake_tfmini/sensor_position/x: 0.0
 * /mavros/distance_sensor/benewake_tfmini/sensor_position/y: 0.0
 * /mavros/distance_sensor/benewake_tfmini/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/mocap_transform: True
 * /mavros/fake_gps/satellites_visible: 5
 * /mavros/fake_gps/tf/child_frame_id: fix
 * /mavros/fake_gps/tf/frame_id: map
 * /mavros/fake_gps/tf/listen: False
 * /mavros/fake_gps/tf/rate_limit: 10.0
 * /mavros/fake_gps/tf/send: False
 * /mavros/fake_gps/use_mocap: True
 * /mavros/fake_gps/use_vision: False
 * /mavros/fcu_protocol: v2.0
 * /mavros/fcu_url: /dev/ttyACM0:921600
 * /mavros/gcs_url: udp://:14551@192....
 * /mavros/global_position/child_frame_id: base_link
 * /mavros/global_position/frame_id: map
 * /mavros/global_position/gps_uere: 1.0
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/tf/child_frame_id: base_link
 * /mavros/global_position/tf/frame_id: map
[... 301 lines cut ...]
[ INFO] [1553990184.063449027]: RC_CHANNELS message detected!
[ INFO] [1553990184.069300839]: IMU: High resolution IMU detected!
[ INFO] [1553990184.074625152]: IMU: Attitude quaternion IMU detected!
[ WARN] [1553990184.083037012]: CMD: Unexpected command 519, result 0
[ INFO] [1553990184.305527794]: CON: Got HEARTBEAT, connected. FCU: PX4 Autopilot
[ INFO] [1553990184.318998914]: IMU: High resolution IMU detected!
[ INFO] [1553990184.320494380]: IMU: Attitude quaternion IMU detected!
[ INFO] [1553990184.360164358]: RC_CHANNELS message detected!
[ INFO] [1553990184.956961940]: FCU: data link #0 lost
[ INFO] [1553990185.073399743]: FCU: data link #0 regained
[ INFO] [1553990185.317123558]: VER: 1.1: Capabilities         0x000000000000e4ef
[ INFO] [1553990185.318149389]: VER: 1.1: Flight software:     01090080 (84c52439e97a4f56)
[ INFO] [1553990185.318947044]: VER: 1.1: Middleware software: 01090080 (84c52439e97a4f56)
[ INFO] [1553990185.319941261]: VER: 1.1: OS software:         071c00ff (1f2f6d8cbe7c85f3)
[ INFO] [1553990185.320979123]: VER: 1.1: Board hardware:      00000011
[ INFO] [1553990185.321760320]: VER: 1.1: VID/PID:             26ac:0011
[ INFO] [1553990185.322598391]: VER: 1.1: UID:                 3138510e35363631
[ INFO] [1553990194.306906521]: HP: requesting home position
[ INFO] [1553990199.311869640]: WP: mission received
[ INFO] [1553990204.307054748]: HP: requesting home position
[ INFO] [1553990214.307091902]: HP: requesting home position
[ INFO] [1553990224.306481161]: HP: requesting home position
```

I am not exactly sure why it crashes. Also, is the trajectory_publisher itself supposed to switch PX4 into the Offboard mode? Or do I need to hold some RC stick in a particular configuration to achieve the switch into Offboard mode?

BhavyanshM, 2019-03-31:
Sometimes the ```trajectory_publisher``` does launch successfully, but the PX4 never switches to Offboard mode. Let me know if you have any suggestions. Even when I have the quadcopter armed and the mode switch set to the Offboard mode, even then the trajectory_planner doesn't have any effect on the PX4.

Jaeyoung-Lim, 2019-03-31:
Have you checked your connection to the flight controller?

Jaeyoung-Lim, 2019-03-31:
@BhavyanshM I have made separate launchfiles for the real vehicle and the SITL simulation to make your life much easier.

You need to set the `fcu_url` correctly, on the mavros side and also set the baudrate according to that on the px4 side in order to make the rpi and mavros be able to talk to each other over the serial.

Hope this works for you

BhavyanshM, 2019-04-01:
> Have you checked your connection to the flight controller? Have you set the `fcu_url` correctly?
> 
> Oh, looking at the logs again you need to comment out rviz too. I was meant to make a launch file for hardware platforms but should have done this earlier
> 
> geometric controller is capable of arming and setting offboard mode automatically but this is intended to be only for simulation
> 
> I propose you assign it to the transmitter switch. trajectory publisher will not do this automatically

Yes, I did have the ```fcu_url``` set properly and was connected to the Pixhawk over Mavros. As indicated by ```[ INFO] [1553990184.305527794]: CON: Got HEARTBEAT, connected. FCU: PX4 Autopilot``` in the logs above. 

I also have a three-way mode switch configured to modes ```Stabilize, AltHold and Offboard```, however, on switching the Offboard mode from the switch nothing happens, and the Pixhawk remains in the current mode and does not switch to the Offboard mode. 

I am published a PoseStamped message at ```/mavros/vision_pose/pose ```, where position.x, position.y and orientation.z are taken from the 2D LIDAR SLAM (Cartographer), position.z is taken from the rangefinder (TFMini), orientation.x and orientation.y are taken from the /mavros/local_position topics. Do you think this is not the right approach or do you think the /mavros/vision_pose/pose is not the right topic to be publishing the quadcopter pose at? I am a bit puzzled at why the Offboard mode is not willing to cooperate now. I really need to get this working in the next one week. Could I please contact you somehow? I understand that you probably have a lot of other things to do. However, I would be really really greatful to you if you could please help us out. It's a matter of an entire team of students. I hope you understand our position. Thanks!

Jaeyoung-Lim, 2019-04-08:
Closing, as this is a usage question

closed 2019-04-08 by Jaeyoung-Lim