Jaeyoung-Lim/mavros_controllers #78: Trajectory Plot and Error plot

shivamgarg1901, 2019-04-13:
Hi! Jaeyoung-Lim 
I was trying to plot the Trajectory using rqt_plot but it is not taking the message mavros/PoseMessage/pose. Please help me out with same.
Also, can you tell me any other method how could I plot the deviation of actual trajectory from the desired trajectory?

Jaeyoung-Lim, 2019-04-13:
@shivamgarg1901 Could you explain more on what is not working with rviz?

On the deviation of the actual trajectory to the desired trajectory, is not being visualized yet. I will try to implement this when I have time. 

I am also trying to measure the statistics of the error the controller has in order to measure the performance of each controller on #77 

Hope this helps

Jaeyoung-Lim, 2019-04-13:
@shivamgarg1901 Actually, it wasn't too hard to implement it so I made a PR on #79 

Please feel free to test it out

closed 2019-04-15 by Jaeyoung-Lim

shivamgarg1901, 2019-04-19:
Hi! 
Thanks for your reply. I tried to run the updated repository.It is running everything except it is not showing any drone in the gazebo environment and not showing any trajectory in Rviz. Here is the screenshot of the error it is showing.
[image]
Please help me out with the same.

PR #79 (2019-04-15): Publish mav path
[image]

This PR publishes the path of the actual mav within a given time window frame. This window size can be adjusted with `/geometric_controller/posehistory_window` and the default is 200 time steps (2.0 seconds)

This Fixes #78