Jaeyoung-Lim/mavros_controllers #80: Error while running the updated repository

shivamgarg1901, 2019-04-19:
Hi! 
I tried to run the updated repository.It is  running everything except it is not showing any drone in the gazebo  environment and not showing any trajectory in Rviz. Here is the  screenshot of the error it is showing. 
[image]
 Please help me out with the same.

Jaeyoung-Lim, 2019-04-19:
Did you rebuild everything? Does it work if you revert the repo?

shivamgarg1901, 2019-04-19:
Yes, It worked fine now. Thanks for your help.
Also, Jaeyoung I launched the controller benchmark file and I cannot find error calculating functionality in it. It is similar to the launching of the sitl_trajectory_track_circle.launch file.

Jaeyoung-Lim, 2019-04-19:
@shivamgarg1901 You need to look at the code. It is not publishing any metrics yet.

shivamgarg1901, 2019-04-19:
Sorry, My Bad. I checked it now. So how could I publish the metric or should I simply add the print statement in code to print it on the terminal?

Jaeyoung-Lim, 2019-04-19:
@shivamgarg1901 You could do that, or you can simply publish the error metrics.

I haven't had time to do that yet, but if you do so, also a contribution will be also welcome :smile:

closed 2020-10-12 by Jaeyoung-Lim