Jaeyoung-Lim/mavros_controllers #82: Iris is making a flip

FaboNo, 2019-04-29:
Hi Jeayoung,

Thank you for sharing the code of the geometric controller. Actually I tried to run the controller alone and when I took a look at the code, I understood that the drone takes off and then it is hovering at Z = 2, waiting for a new command.

So I entered `roslaunch px4 mavros_posix_sitl.launch` in order to start gazebo, px4 and Mavros
and `roslaunch geometric_controller trajectory_controller.launch`

I slightly modified the launch file as follow (I just kept the controller along with the required parameters):

```
<launch>
  <arg name="mav_name" default="iris"/>
  <arg name="command_input" default="2" />
  <arg name="gazebo_simulation" default="true" />
  <arg name="visualization" default="true"/>
  <arg name="log_output" default="screen" />
  <arg name="respawn_mavros" default="false" />

  <node pkg="geometric_controller" type="geometric_controller_node" name="geometric_controller" output="screen">
  		<param name="mav_name" type="string" value="$(arg mav_name)" />
          <remap from="command/bodyrate_command" to="/mavros/setpoint_raw/attitude"/>
          <param name="ctrl_mode" value="$(arg command_input)" />
          <param name="enable_sim" value="$(arg gazebo_simulation)" />
          <param name="enable_gazebo_state" value="true"/>
  </node>
</launch>
```
What I observe is that it is take off and when it reached the target pose, it makes a flip and crashed few meters away... I have a log file with the values of` q_des, cmdBodyRate_` and so on but before digging into that file, I would like to know if I missed something ?

Thank you for your help

Regards

Fabrice

Jaeyoung-Lim, 2019-04-29:
@FaboNo Thank you for taking interest of the code. The sequence of the startup is something that I have not tested. Could you instead try and run

```
roslaunch mavros_controllers sitl_track_trajectory.launch
```

FaboNo, 2019-04-30:
@Jaeyoung-Lim I tried
`roslaunch geometric_controller sitl_trajectory_track_circle.launch`
It starts sitl and I can see on rviz, a point circling but the behavior of the drone was the same as before: take off, reach the target, flip and fall apart.
I guess probably something is wrong in my setup? Actually the only difference is with eigen, in the include files (geometric_controller.h) I had to add eigen3 to /eigen/Dense.h 
I will continue the investigation...
Thank you for your help
Fabrice

Jaeyoung-Lim, 2019-04-30:
Now I see the problem. Sorry for not being clear on the documentation

You are using the mavros release. All the development was done on the current mavros source builds. This is due to some of the missing functionalities of the current release.

The crash is likely because unfortunately some of the topics are empty.

FaboNo, 2019-04-30:
Hi Jaeyoung,

You mean I have to clone the current version of mavros (https://github.com/mavlink/mavros) ?

Fabrice

Jaeyoung-Lim, 2019-04-30:
Exactly

FaboNo, 2019-04-30:
Yes it works fine now!
Ok next step follow a trajectory that I will provide
Thank you for your help Jeayoung

closed 2019-04-30 by FaboNo

Jaeyoung-Lim, 2019-04-30:
@FaboNo awesome! the interface is already there through MultiDOFJointTrajectory messages. hope this is useful for you

FaboNo, 2019-04-30:
@Jaeyoung-Lim Yes I saw it, but in the geometric controller, only the indice 0 is taking into account:
```
targetPos_ << pt.transforms[0].translation.x, pt.transforms[0].translation.y, pt.transforms[0].translation.z;
targetVel_ << pt.velocities[0].linear.x, pt.velocities[0].linear.y, pt.velocities[0].linear.z;
```
Besides if I do not provide the acceleration, does it matter?

Otherwise, I can use the `reference/setpoint` interface if I am able to update the ref point each xxxms

Thank you for sharing this nice piece of work!

Jaeyoung-Lim, 2019-04-30:
@FaboNo Thanks! 

Providing acceleration gives you better performance to high acceleration trajectories, but for normal trajectories it doesn't make much of a difference.

The plan was to have the trajectoryPublisher publish reference along trajectories along sampled points as you mentioned. I believe the controller should stay as a reference tracker, while a separate note gives references along the trajectory.

The problem I was having was that I didn't have a good example to test this. The shape trajectories was sort of the preliminary work for such functionality.

If you can contribute such functionality it would be super awesome!

FaboNo, 2019-04-30:
@Jaeyoung-Lim Sure if I can contribute I will! Let me see how to do and get back to you as soon as I have a viable solution