Jaeyoung-Lim/mavros_controllers #83: thrust_scaling parameter is set to zero

altshuler, 2019-05-12:
Hi,

After everything worked fine on SITL I modified launch file in order to run it on real flight.
I disabled gazebo_simulation ,as you recommended,  and  tried to run  geometric_controller using  **roslaunch geometric_controller sitl_trajectory_track_circle.launch**    command  on real 
vehicle, but I get  this warning:
[WARN] [1557678319.767286131]: thrust_scaling parameter is set to zero.

I can change vehicle state to OFFBOARD and arm, but it's not taking off.
I figured out that it's    thrust_scaling  parameter should be 1.0.
Where and how I can modify this parameter ?

Jaeyoung-Lim, 2019-05-13:
@altshuler Are you running current master?

If so, have you also tried 
```
roslaunch geometric_controller mav_trajectory_track_circle.launch
```

altshuler, 2019-05-13:
@Jaeyoung-Lim 
the mavros is build from source, maybe not the latest version...
I will check it. What version you are using ?

Jaeyoung-Lim, 2019-05-13:
@altshuler The launchfile I suggested was created specifically for real vehicles.

I am not sure what modifications you have made in the launchfile, but that might have been causing the problem.

altshuler, 2019-05-13:
ok, I will test it,
but I think its mavros issue, because simulation I'm running 
on PC and mavros is built from source, while on vehicle also mavros is built 
from source, but little bit outdated. First of all I will update mavros on vehicle.

altshuler, 2019-05-15:
Yesterday I tried to fly on real vehicle.
It's looks good. One thing I noticed that is 
takeoff  is very aggressive.  First  vehicle flies aside with a huge tilt.
And after it locked on trajectory it files fine. Same behavior also seen in 
simulation.

Jaeyoung-Lim, 2019-05-15:
@altshuler Cool! Do you have any video from the flight? 

As for the aggressive take off. You can tune down the gains. The default gains are quite aggressive, as I have put it on the limit. Also, the thrust scaling might be different, as IRIS is not a powerful drone, therefore the thrust scale is small. This might result in high thrust commands on your vehicle if it is more powerful

altshuler, 2019-05-15:
I have no video, but I can do it next flight.
What gain exactly I need to tune ? 
Its not vertical speed I need to decrease ?

altshuler, 2019-05-18:
Today I tested again the circle trajectory.
Here is the link to the video:[https://drive.google.com/open?id=1lc1xgVjS0K7seMkbNHlb-GwRLztKTXVB](url)
and flight log:[https://logs.px4.io/plot_app?log=d7c82011-4305-44b5-92de-d7983ad98e9c](url)
Trajectory tracking seems good.
What gain you suggest to change in order to smooth the takeoff ?

Also I have some altitude slow floating. I think it's not directly related to this test,
but to the PID tuning. Can you please look to the log and advice what parameters I need to tune 
to decrease altitude floating?

Jaeyoung-Lim, 2019-05-19:
@altshuler Thanks! Video looks awesome

So a few things you can tune:

- lower `max_fb_acc` : part of the reason the take off is aggressive is because the error is initially large. You can lower the maximum acceleration that is generated with the error with this parameter. Try tuning down the `Kp_z` and `Kv_z` values so that you get a smoother flight. Seeing that you have oscillations in altitudes, this `Kp_z` might be a little bit high or `Kv_z` a little bit low

- As offboard thrust commands are normalized (-1, +1) `normalizedthrust_constant` defines how to scale your thrust commands to the real drone. This is set to match the IRIS in SITL so you might need to change this value. From the flight, this doesn't look too bad but will still optimize the performance

altshuler, 2019-05-19:
@Jaeyoung-Lim , thank you for the answer.
The altitude oscillation happens not only when I running this code.
It happens in all modes, so maybe first I need to tune PX4 parameters like MPC_Z_VEL_I  or 
MPC_Z_P  ?

Jaeyoung-Lim, 2019-05-19:
@altshuler No, as we are sending bodyrate and thrust commands directly to the firmware, we are not using the position controller which the MPC_ parameters define

altshuler, 2019-05-19:
Ok, thanks. I will test it and update you after my next flight test.

altshuler, 2019-05-26:
I'm flied yesterday, changed **Kp_z** from 10 to 25 and  **Kv_z** from 3.3 to 2.3, **max_fb_acc_**= 5.0.
It seems no effect - same altitude floating. Maybe it's will be useful  to change **MPC_Z_P, MPC_Z_VEL_D, MPC_Z_VEL_I, MPC_Z_VEL_P** parameters, to see if it possible to tune, because I have  same behavior in  position mode .

Jaeyoung-Lim, 2019-05-27:
Could you clarify on which launchfiles you are running and where you are changing the parameters?

altshuler, 2019-05-27:
I'm understand that, but if it's impossible to tune MPC params is this can point to some other problem?
I'm running geometric_controller with  sitl_trajectory_track_circle.launch file. I disabled simulation and inserted all Kp_ , Kv_ and max_fb_acc  in this launch file under geometric_controller configurations.

Jaeyoung-Lim, 2019-05-27:
@altshuler That is strange. Do you see any behavior change in SITL?

Jaeyoung-Lim, 2019-05-27:
@altshuler I think I found the problem:
https://github.com/Jaeyoung-Lim/mavros_controllers/blob/87889e869e524259503d8e1e0a189fd32a3c399b/geometric_controller/launch/sitl_trajectory_track_circle.launch#L27

The parameters should be under geometric controller not trajectory publisher.

altshuler, 2019-05-27:
@Jaeyoung-Lim , Kpos_x,Kpos_y, Kpos_z are not used in geometric_controller.
As I changed the sitl_trajectory_track_circle.launch I can see changes in simulation when
I change Kp_z , Kv_z.

Jaeyoung-Lim, 2019-05-27:
@altshuler That is parameter mismanagement on my side. I will make a fix.

juanmed, 2019-05-31:
@Jaeyoung-Lim  could you please confirm the last tested version of mavros?  

> @altshuler The launchfile I suggested was created specifically for real vehicles.
> 
> I am not sure what modifications you have made in the launchfile, but that might have been causing the problem.
> 
> I will get back to you with the version last tested, but it is quite recent. It is unfortunate that this project doesn't work on the latest release, but it is because of a regression introduced from the mavros side

Jaeyoung-Lim, 2019-05-31:
@juanmed The current upstream mavros works with this package. You need to build mavros from source

juanmed, 2019-05-31:
@Jaeyoung-Lim Thanks for the confirmation. I built latest mavros from source. I posted an issue on #87, I thought it might be due to mavros but maybe not. I will appreciate if you take a look at it. Thanks.

juanmed, 2019-06-06:
Hello, I just wanted to post a link to a video using mavros_controllers on a real drone. Like @altshuler , I am using an Intel Aero for this flight:

https://www.youtube.com/watch?v=6Yaq4uRLL6E

I launched with  the mav_trajectory_track_circle.launch:

roslaunch geometric_controller mav_trajectory_track_circle.launch fcu_url:="tcp://:5760@127.0.0.1:5760"

Jaeyoung-Lim, 2019-06-06:
@juanmed Thank you for the video! Really cool to see it working!

altshuler, 2019-06-10:
@Jaeyoung-Lim  , I'm added dynamic reconfigure option in geometric_controller to be able configuring PID 
values in simulation and real flights. If it's ok I can put PR.

Jaeyoung-Lim, 2019-06-10:
@altsuler That would be awesome! Contributions are always welcome :+1:

closed 2020-10-12 by Jaeyoung-Lim