Jaeyoung-Lim/mavros_controllers #92: Compilation error :geometricCtrlConfig.h: No such file or directory

germal, 2019-06-13:
Hello

Hi !
I just found the following compilation error  :

In file included from /home/ubuntu/catkin_ws/src/mavros_controllers/geometric_controller/src/geometric_controller.cpp:3:0:
/home/ubuntu/catkin_ws/src/mavros_controllers/geometric_controller/include/geometric_controller/geometric_controller.h:34:54: fatal error: geometric_controller/geometricCtrlConfig.h: No such file or directory

and ìndeed the header file is not present .
Clould you please advice ?
Thanks a lot !

[image]

Jaeyoung-Lim, 2019-06-13:
@germal It seems like a regression from #90 . Could you revert the PR for the moment? I am working on a fix.

Jaeyoung-Lim, 2019-06-13:
@germal Could you try again with the current master? I believe the issue is fixed

germal, 2019-06-13:
@Jaeyoung-Lim  , it compiles perfectly ! Thank you very much !

closed 2019-10-28 by germal