Jaeyoung-Lim/mavros_controllers #96: Where is the reference position and acceleration and jerk?

tuloski, 2019-07-04:
I don't see where the position and acceleration and jerk trajectory is taken since the only topic it's subscribed is a geometry_msgs/TwistStamped. 
And they are needed by the controller for both feedback (position) and feedforward (acceleration and jerk) terms.
Also the trajectory_publisher seems only to generate a geometry_msgs/TwistStamped setpoint.

Is the position integrated from the velocity (which is btw not really correct)?

tuloski, 2019-07-04:
Ok, looking at the code I saw that the position is taken from the twist angular (ouch!).
A little info about this in the README would be nice.

I also see there is a not documented controller_msgs::FlatTarget message which can be used to send the complete trajectory up to snap which is nice but it is not documented.

Jaeyoung-Lim, 2019-07-06:
@tuloski Thanks for the feedback. The twist input was a hack that I put together in the initial stage of the project. I will make a fix on it

closed 2021-03-29 by Jaeyoung-Lim