Jaeyoung-Lim/mavros_controllers #99: How to make geometric controller to track circle trajectory

SwiftGust, 2019-07-22:
[image]

Hello, 
I tried to make put iris to track circle trajectory more perfectly, but could not make vehicle to trakc more closely.
How can I make iris to track circle or any trajectory more closely? 

It looks like only these parameters are available to the user
```
  nh_private_.param<string>("mavname", mav_name_, "iris");
  nh_private_.param<int>("ctrl_mode", ctrl_mode_, MODE_BODYRATE);
  nh_private_.param<bool>("enable_sim", sim_enable_, true);
  nh_private_.param<bool>("velocity_yaw", velocity_yaw_, false);
  nh_private_.param<double>("max_acc", max_fb_acc_, 7.0);
  nh_private_.param<double>("yaw_heading", mavYaw_, 0.0);
  nh_private_.param<double>("drag_dx", dx_, 0.0);
  nh_private_.param<double>("drag_dy", dy_, 0.0);
  nh_private_.param<double>("drag_dz", dz_, 0.0);
  nh_private_.param<double>("attctrl_constant", attctrl_tau_, 0.1);
  nh_private_.param<double>("normalizedthrust_constant", norm_thrust_const_, 0.05); // 1 / max acceleration
  nh_private_.param<double>("Kp_x", Kpos_x_, 8.0);
  nh_private_.param<double>("Kp_y", Kpos_y_, 8.0);
  nh_private_.param<double>("Kp_z", Kpos_z_, 10.0);
  nh_private_.param<double>("Kv_x", Kvel_x_, 1.5);
  nh_private_.param<double>("Kv_y", Kvel_y_, 1.5);
  nh_private_.param<double>("Kv_z", Kvel_z_, 3.3);
  nh_private_.param<int>("posehistory_window", posehistory_window_, 200);
```

And is dob controller package working?
With dhat min & max = 0.0 and other parameters left as default, iris cannot get close to circle trajectory and always end-up crashing.

Jaeyoung-Lim, 2019-07-22:
@SwiftGust By more closely, if you mean by position error only you need to tune down the gains of the velocity.(`Kv_*`) The current control error is defined as the weighted sum of position and velocity.

Tracking position perfectly is not a good thing especially in trajectories with high acceleration, as it will not be able to follow the setpoint's velocity changes. Therefore a feedforward gain of velocity and optionally acceleration is used for the controller reference

Jaeyoung-Lim, 2019-07-22:
@SwiftGust The dob package is also working. Did you see any problems with the package?

SwiftGust, 2019-07-22:
@Jaeyoung-Lim 
I got the point but, I thought this package also has capability of tracking such trajectory as like one of reference 

> https://www.youtube.com/watch?v=VIQILwcM5PA&feature=youtu.be

> Faessler, Matthias, Antonio Franchi, and Davide Scaramuzza. "Differential flatness of quadrotor dynamics subject to rotor drag for accurate tracking of high-speed trajectories." IEEE Robot. Autom. Lett 3.2 (2018): 620-626. 

About dob, I thought by setting d_hat to be 0, maybe able to make sure dob package running without estimated disturbance affecting the outer loop controller, so that help me to point out either outer-loop controller or dob controller has problem.

I tried again with all following parameters 0
```  
  nh_.param<double>("/dob_controller/dob/a0_x", a0_x, 0.0);
  nh_.param<double>("/dob_controller/dob/a0_y", a0_y, 0.0);
  nh_.param<double>("/dob_controller/dob/a0_z", a0_z, 0.0);
  nh_.param<double>("/dob_controller/dob/a1_x", a1_x, 0.0);
  nh_.param<double>("/dob_controller/dob/a1_y", a1_y, 0.0);
  nh_.param<double>("/dob_controller/dob/a1_z", a1_z, 0.0);
  nh_.param<double>("/dob_controller/dob/a1_y", a1_y, 0.0);
  nh_.param<double>("/dob_controller/dob/a1_z", a1_z, 0.0);
  nh_.param<double>("/dob_controller/dob/tau_x", tau_x, 0.0);
  nh_.param<double>("/dob_controller/dob/tau_y", tau_y, 0.0);
  nh_.param<double>("/dob_controller/dob/tau_z", tau_z, 0.0);
  nh_.param<double>("/dob_controller/dob/max_dhat", dhat_max, 0.0);
  nh_.param<double>("/dob_controller/dob/min_dhat", dhat_min, 0.0);
```

Vehicle is not even taking off due to orientation in topic reference/pose contains NaNs

```header: 
  seq: 4193
  stamp: 
    secs: 45
    nsecs:  72000000
  frame_id: "map"
pose: 
  position: 
    x: -1.57392626171
    y: -1.23400004972
    z: 2.0
  orientation: 
    x: nan
    y: nan
    z: nan
    w: nan
---
```

Jaeyoung-Lim, 2019-07-22:
@SwiftGust It is the same implementation.

As you can see in the video you have cited, the controller is unable to track the velocity properly without drag compensation. You can tune the drag to follow the performance as in the video. The drag coefficients are set to zero in default.

Can you fly with dob with the default parameters or not? If you make everything zero, you will be dividing numbers into zero according to the current implementation. https://github.com/Jaeyoung-Lim/mavros_controllers/blob/master/dob_controller/src/dob_controller.cpp#L91

SwiftGust, 2019-07-23:
@Jaeyoung-Lim Thank you for detailed explanation.
What I want to achieve is fast, unnoticeable close response to the reference input without position deviation from actual reference input, as with drag compensation in the video.

For the DOB, it cannot fly with default parameters in master branch, ends-up crashing so that's why I want to find out what is causing the problem.
From your explanation, the dob controller implemented here, it does affects reference/pose regardless of it returns control input for disturbance rejection.
Then, it would not be able to isolate problem

SwiftGust, 2019-07-23:
Additionally, it is also hard to make track figure-8 trajectory with some gain tuning

Here attached gain & launch file to reproduce issue (*.txt format for uploading).
[sitl_trajectory_track_circle.txt](https://github.com/Jaeyoung-Lim/mavros_controllers/files/3420290/sitl_trajectory_track_circle.txt)
[figure-8-failed.txt](https://github.com/Jaeyoung-Lim/mavros_controllers/files/3420292/figure-8-failed.txt)

Jaeyoung-Lim, 2019-07-23:
@SwiftGust Seems like there is no velocity reference for lemniscate
https://github.com/Jaeyoung-Lim/mavros_controllers/blob/a28e89adbc9a7470577eb04d80d6d84a6e023576/trajectory_publisher/src/shapetrajectory.cpp#L91

Jaeyoung-Lim, 2019-08-07:
@SwiftGust Any updates ?

SwiftGust, 2019-08-08:
Sorry for delayed update, after testing PR #100 and talk, 
caught up with other tasks so no further progress yet.
Will open another issue if there is problem, thank you.

closed 2019-08-08 by SwiftGust