Jaeyoung-Lim/mavros_controllers #101: What are the Equations Used in This Implementaion?

mzahana, 2019-08-19:
Hi @Jaeyoung-Lim 

I am trying to find which equations are used to implement the controller in this packahe. In particular, in "Geometric tracking control of a quadrotor UAV on SE (3)" The inputs are collective thrust force, f, and moments M. I beleive it is simple to send f to mavros setpoint. However,  how is the computed moments input, M, mapped to attitude rate setpoints?
Also the, computatin of M requires the knoledge of the inertia matrix I, which I don't think is implemented in this package, right? If so, how is that handled?

Thanks in advance.

Jaeyoung-Lim, 2019-08-19:
@mzahana I explicitly avoided using force and moments for that reason. The attitude controller is something similar from https://www.research-collection.ethz.ch/bitstream/handle/20.500.11850/154099/eth-7387-01.pdf

mzahana, 2019-08-19:
Thanks, @Jaeyoung-Lim . So, the implementation of the controller in this package is based on the report you mentioned, or there are other reports/papers?

Jaeyoung-Lim, 2019-08-19:
@mzahana Only the attitude controller is based on the link I provided. The rest is based on the papers on the readme.

However, there is no single paper that this implementation points to. The implementation was done carefully to get the best out of controlling the drone aggressively with mavros.

mzahana, 2019-08-31:
Thanks @Jaeyoung-Lim . Do you know of packages for trajectory generations for quadrotors? I am aware of several techniques, but didn’t see an implementation for that.

Jaeyoung-Lim, 2019-08-31:
@mzahana There are actually quite a lot around. What applicationsare you looking for? I can recommend you a few

mzahana, 2019-08-31:
I would like to generate fastest feasible position trajectory that goes through some waypoints with re-planning in case of popping obstacles.

Jaeyoung-Lim, 2019-08-31:
@mzahana Have you looked at [mav_trajectory_generation](https://github.com/ethz-asl/mav_trajectory_generation) ?

mzahana, 2019-08-31:
@Jaeyoung-Lim thanks!
Looks like a good start. Have you used that with this package (mavros_controllers)?

Jaeyoung-Lim, 2019-08-31:
@mzahana I haven't tested it personally, but we have a interface on the MultiDOFJointTrajectory Message as one of the contributors was using it with the geometric_controller.

Robby119, 2020-06-20:
> @mzahana I explicitly avoided using force and moments for that reason. Moreover, it is not safe to have actuator inputs as a offboard command as the serial connection is prone to packet loss and connection breakages.
> 
> The attitude controller is something similar from https://www.research-collection.ethz.ch/bitstream/handle/20.500.11850/154099/eth-7387-01.pdf

Hi, i'm still confused about why attitude rate as an input is safer than actuator input in offboard mode, is it related to part of controller that being skip in px4 when using a actuator input?

closed 2021-03-29 by Jaeyoung-Lim