Jaeyoung-Lim/mavros_controllers #102: Kp_z value

juanmed, 2019-08-24:
Hi,

The `Kp_z` gain value is assigned as  `30` in the launch files. 

https://github.com/Jaeyoung-Lim/mavros_controllers/blob/6ae4ef07e601f4f2a3fb334b44991f4744a613ec/geometric_controller/launch/sitl_trajectory_track_circle.launch#L23

https://github.com/Jaeyoung-Lim/mavros_controllers/blob/6ae4ef07e601f4f2a3fb334b44991f4744a613ec/geometric_controller/launch/mav_trajectory_track_circle.launch#L28  (A bug though, In this launch file, these values should be under the `geometric_controller` node instead of the `trajectory_publisher`)

I have been doing various test with mavros_controllers and found that this value causes the drone to crash in simulation. However, the default value of `10` does not make the controller unstable and the drone can track the trajectory. I was wondering:

- Have you tried the value `Kp_z = 30` in a real drone? 

I have tried the value `Kp_z = 10` in an Intel Aero RTF and the flight was stable, but I do not want to try the value of 30 after what happens in simulation. See [this video](https://www.youtube.com/watch?v=_m9JE7PGKwo) where I show the performance of both values in simulation.

I came to notice this since a controller I am developing works in simulation but not on the real drone (Intel Aero RTF). However, mavros_controllers works both in sim and real world, in the same drone. I wanted to ask, do you have any ideas why if both controllers work in sim only mavros_controllers work in the real drone?

Thanks!

Jaeyoung-Lim, 2019-08-24:
@juanmed Thanks for the comments. Does this mean that on the master branch the drone is unable to track the trajectory? I have not tried `Kp_z = 30` on a real drone.

`Kp_z` is one of the constants that depends highly on the actual hardware. The reason is that we command thrust as normalized thrust values but the actual thrust generated from these normalized thrust values can vary greatly depending on the hardware.

Could you explain more about your controller? The reason the behavior of `geometric_controller` is simular between hardware and simulation is that the controller is sending control inputs that are not dependent on the hardware. For example the attitude rate commands are part of the feedback loop of the flight controller, therefore the behavior will be similar. This was a design choice to make it as similar as possible.

juanmed, 2019-08-25:
> Does this mean that on the master branch the drone is unable to track the trajectory?

Yes, at least in simulation in my computer. I did not tried on the real drone.

Thanks for the explanation on `Kp_z`. Regarding my controller, just like yours it sends thrust and angular velocity commands to the `/mavros/setpoint_raw/attitude` topic. The values are calculated following using the expressions in your Reference [1], both for the thrust and the desired body-frame angular velocities. So the calculation of thrust is very similar to your geometric_controller, but angular velocity is completely different. Trajectories are generated using differential flatness but some the mass used is that for iris drone `1.5`kg. This works in simulation but not on the real drone.

Jaeyoung-Lim, 2019-08-25:
@juanmed Okay, in that case I would appreciate if you make a fix on the launchfile default gains?

Have you checked if your controller is sending realistic values in terms of jerk when sending angular rates? It might be that your vehicle is simply unable to track the control inputs you are sending.

closed 2021-01-31 by Jaeyoung-Lim