Jaeyoung-Lim/mavros_controllers #103: IDE question

altshuler, 2019-08-25:
Hi, @Jaeyoung-Lim 

I have a question about IDE.
I debugged mavros_controllers with eclipse.
My steps as following:
1. Running geometric_controller from terminal using roslaunch geometric_controller        sitl_trajectory_track_circle.launch
2. launching eclipse and debugging the code.

In this case previously running node is suspended as another node with the same name is launched. 
But after last commit its stopped working. 

My intention is to run the launch file directly from IDE and then debug.
I found that QT creator with ROS plugin can do it, But unfortunately I have 
some problems with QT creator configurations. 

What IDE you are using and what the steps for running and debugging the code ?

Jaeyoung-Lim, 2019-08-25:
@altshuler I use visual studio code as I find it quite convenient, as well as it is well integrated with px4 firmware.

For debugging I usually run gdb or valgrind in the prefix as explained [here](http://wiki.ros.org/roslaunch/Tutorials/Roslaunch%20Nodes%20in%20Valgrind%20or%20GDB)

altshuler, 2019-08-25:
@Jaeyoung-Lim , thank you for your answer.
Is it possible to open whole catkin_ws workspace in visual studio code in order to see all projects?
 Is there any add-ons needed to be installed for PX4 support ?
Anyway, as I understand ,the sequence is first running launch file with gdb prefix added, then start debug session in IDE ?

altshuler, 2019-09-02:
I'm installed Visual studio code ,ROS plugin and C++ plugin.
Also added to sitl_trajectory_track_circle.launch file gdb prefix:
`<node pkg="geometric_controller" type="geometric_controller_node" name="geometric_controller" output="screen" launch-prefix="gdb --args">`
Unfortunately I'm still not able to debug  using breakpoints in VS code.
My steps are : 
1. launching  sitl_trajectory_track_circle.launch  file from external terminal
2. Start debug session in VS code.

As I start debug session another terminal is opened automatically in VS code and node is opened.
But the drone stays on the ground and not arming.

Jaeyoung-Lim, 2019-09-02:
@altshuler Is this specific to this package? or is this something happening in ROS in general?

altshuler, 2019-09-02:
I don't know yet. I just tried to run another package, not using .launch file.
Generally its running. 
Today will try to run another package using .launch file.

altshuler, 2019-09-02:
What is your steps for starting debug session in VS code ?
the prefix in launch file should include "run" option as:
`launch-prefix="gdb  run --args"`    ?

closed 2021-03-29 by Jaeyoung-Lim