Jaeyoung-Lim/mavros_controllers #110: Where is the trajectory published to?

HenryHuYu, 2019-12-27:
Hi thanks for your great work, I got 2 questions:

1. I saw that the trajectory publisher has a varieble 
```
 trajectoryPub_ = nh_.advertise<nav_msgs::Path>("/trajectory_publisher/trajectory", 1);
```
but I didnot find a subsciber that subs this trajectory. Any help with this?

2. To my understanding, your polynomial trajectory is something like 
p(t) = c1 + c2 * t + c2 * t^2 + c3 * t^3 
is that right?

Jaeyoung-Lim, 2019-12-27:
1. the geometric controller takes the reference from the trajectory publisher not the trajectory. 

2. that is correct

HenryHuYu, 2020-01-02:
Thx so much. Here I got another problem that when I try to run your simulation in the tutorial (tracking a circle trajectory). I got the following error, sometimes the mavros controller node will crash, but sometimes it tracked a circle :
[image]

I didnt press any keyboard.

Jaeyoung-Lim, 2020-01-02:
@HenryHuYu It seems like your SITL simulation is crashing and it is not related to this repo. Are you running the process with root previliges? See that your px4 daemon is complaining that the resource is not available.

HenryHuYu, 2020-01-04:
@Jaeyoung-Lim I found out. It is because I run SITL simulation separately while your launch file contains SITL simulation node. 
Here is another question:
```
a_rd = R_ref * D_.asDiagonal() * R_ref.transpose() * targetVel_ 
```
but I found that D_ is always 0. I checked the paper "Automatic Re-Initialization and Failure Recovery for Aggressive Flight with a Monocular Vision-Based Quadrotor" but found out that the position controller is something like(without a rotor drag): 
```
a des = P pos ·(r ref − r̂)+D pos ·(v ref − v̂)+a ref −g
```
So what is **Rotor Drag** ? Is there any related paper? 
Thx very much!

HenryHuYu, 2020-01-06:
@Jaeyoung-Lim Also, is the function in geometric_controller.cpp ever used?
```
void geometricCtrl::multiDOFJointCallback(const trajectory_msgs::MultiDOFJointTrajectory& msg) 
```

Jaeyoung-Lim, 2021-03-29:
@HenryHuYu It is used when you publish to the topic that triggers the callback

closed 2021-03-29 by Jaeyoung-Lim