Jaeyoung-Lim/mavros_controllers #111: Design question: why store quaternions in Vector4d instead of Quaterniond

juanmed, 2020-01-14:
Hi,

I saw you are using `Eigen::Vector4d` for storing the attitude quaternion instead of the `Eigen::Quaterniond` or `Eigen::Quaternionf` data structures. I was wondering if you could share what were the design considerations when making this decision. Thanks.

Jaeyoung-Lim, 2020-01-14:
@juanmed I think using `Eigen::Quaternionf` would be better. We need to change this at some point

Jaeyoung-Lim, 2020-01-14:
@juanmed I will be adding unit tests in the next weeks to make this project more robust.

If you are using this project, it would be cool if we can discuss what are the missing pieces are in this project so that we can make our lives easier :)

juanmed, 2020-01-16:
@Jaeyoung-Lim Hi, yes for sure. Let me make a list of what I consider can be improved and send it to you. Notwithstanding, I will be following your lead. I think I can close this issue now.

closed 2020-01-16 by juanmed