Jaeyoung-Lim/mavros_controllers #114: Sending Yaw Commands

matthew-givens, 2020-02-06:
Hi,

I'm pretty new to ROS (and C++ for that matter). I've gotten everything working in the simulation and I have a real drone set up with position feedback. Now I'm looking to have the simulated drone point at the center of the circle at all times (then I'll test it with the real drone). I know how to generate the appropriate yaw quaternion command (using the theta angle about the circle) but I don't know how to implement this in the code. I've written a 'getOrientation' function in the shapetrajectory.cpp file but this seems to only affect the initial orientation. Any guidance on this would be appreciated!

Thanks!

Jaeyoung-Lim, 2020-02-07:
@matthew-givens Thank you for the nice words. It would be great if you can provide feedback or even better if you can contribute to the code base. 

You can set yaw by simply sending topics over `~reference/yaw` message

matthew-givens, 2020-02-07:
Thanks! With some effort, I wrote a package that subscribes to the `/mavros/local_position/pose` topic, calculates my desired yaw, and publishes Float32 yaw messages to the `/reference/yaw` topic. And it works! 

If I create anything worth sharing I will definitely try to contribute.

closed 2020-10-12 by Jaeyoung-Lim