Jaeyoung-Lim/mavros_controllers #121: setpoint_raw/attitude vs. actuator_controls

dominiknatter, 2020-03-11:
Hi Jaeyoung,

first of all: nice implementation! :)

I'm currently thinking of supporting the PX4 controller by flatness-based feedforward control (to exploit knowledge about the model). For this I digged a bit into the PX4 firmware, to see how this could be implemented. I concluded that sending PWM values is probably a step too far and I should rather send body rates and a thrust value. For me it looked reasonable to use MAVROS' actuator_control message for this. However, your implementation seems to remap the body rates to the setpoint_raw/attitude message instead
https://github.com/Jaeyoung-Lim/mavros_controllers/blob/master/geometric_controller/launch/trajectory_controller.launch#L16

Can you tell me why you chose to do so? What happens to this setpoint? Is it still fed into mc_att_control or mc_rate_control? Or does it directly enther the px4fmu module? In the latter case it sounds like it's identical to actuator_control? I didn't find the necessary information yet, unfortunately.

Thanks in advance for your help,
Dominik

Jaeyoung-Lim, 2020-04-19:
@dominiknatter Awesome project! Yes, the setpoints are set to mc_rate_control so we send bodyrate setpoints. This was a sensible decision to have better tracking performance with aggressive maneuevers.

Usually when running offboard applications, you are communicating through a serial interface with the flight controller which introduces latency and communication bandwidth. Controlling actuator inputs through this link is not really safe and also you will only get limited performance out of it. (usually a few 100Hz command rate)

On the other hand, if you send body rate setpoints, you can assume that the vehicle will follow the bodyrate setpoints instantly. (This assumption have been proved to be valid until quite aggressive flights in multiple papers) The reason that body rate tracking works so well is that the fast rate control loop running in px4(~1kHz) and also IMU reading rates (~3kHz or more) result in a pretty stable feedback loop. 

Does this answer your question?

dominiknatter, 2020-04-20:
@Jaeyoung-Lim thanks for the feedback!

I'll repeat to make sure I got your point: you agree, that using setpoint_raw/attitude does not strictly stand for pure feedforward (FF), as it passes through another controller (mc_rate_control that is). However, this remaining loop reacts so fast that the performance is affected only minimally (if at all) and hence it's practically FF.

Assuming that's correct, this sounds like I should use /mavros/setpoint_raw/attitude (which corresponds to SET_ATTITUDE_TARGET in MAVLink) to provide my FF terms (instead of  /mavros/actuator_controls and SET_ACTUATOR_CONTROL_TARGET). I guess that inherently means I cut off feedback from mc_pos_control and mc_att_control (there's not full controller diagram online unfortunately, but I extrapolate this from [here](https://dev.px4.io/v1.9.0/en/flight_stack/controller_diagrams.html#multicopter-position-controller), where either velocity setpoints or the position controller can be used, but not both). That will in turn lead to drift if I'm right.

So I assume to combine my FF with the PX4 feedback I have to use a separate MAVLink+MAVROS message, and adapt mc_rate_control to add this message to the input coming from mc_att_control? I am curious whether you agree with my thinking here :)

Thanks for your help, really appreciate it!

Jaeyoung-Lim, 2020-04-20:
@dominiknatter Although the rate / attitude setpoints are included in the same message, if you send body rate setpoints it is handled by `mc_rate_control` and if you send attitude setpoints, it is handled by `mav_att_control`. So I think you are confusing the two. 

I don't think you need to get carried away of the implementation details in px4. It is just about whether if you assume the plant dynamics is 1st order vs second order depending on the input you use.

dominiknatter, 2020-04-20:
@Jaeyoung-Lim  I intended to underlay a model equal to (1) - (4) [here](http://rpg.ifi.uzh.ch/docs/RAL18_Faessler.pdf). Using their flatness derivation I should then be able to feedforward omega_dot. But what I meant to say is: pure feedforward will not control position and hence drift due to e.g. model uncertainties. That's why
 - in this paper they close the loop with their own position controller
 - your repo uses a tracking law
 - and I wanted to rely on the PX4 controllers to compensate for these small errors.

And that leads to my concern, that I either did not express clearly or is based on a misunderstanding in my head: So far these PX4 controllers work in a cascade, where (being imprecise) some controller publishes a body rate setpoint to the next controller. If I now feedforward my own body rate setpoint, it feels like the other setpoint is overridden. That would mean the overlying controllers are ignored, hence the drift is still here. Is it just a misunderstanding?

Jaeyoung-Lim, 2020-04-20:
@dominiknatter My implementation is basically following that exact paper. 

I don't think I understand exactly what your concerns are. Even if you have model uncertainties your drift will be bounded due to the stability of the controllers. The other control loops are ignored and you need to run your own position/trajectory controller offboard. Therefore, you cannot use the body rate controller together with the higher level controllers of PX4. This is not how offboard control works.

dominiknatter, 2020-04-20:
@Jaeyoung-Lim ahh now we got it. That last paragraph exactly nailed my concern, so I better tell my supervisor that we shouldn't try to misuse offboard mode in such a way.

We just thought it would be "a waste" to drop the PX4 controllers, as they work quite well :) and we already have a time-optimal trajectory generator that would allow to calculate the flatness-based feedforward terms. But apparently we should think differently.

Thanks a lot for your insights!

closed 2020-04-20 by dominiknatter