Jaeyoung-Lim/mavros_controllers #122: is it usable for hexacopter ?

Robby119, 2020-03-25:
i'm doing my research but using hexacopter, i'm wondering is it usable for hexacopter?  if it isn't what should i change from your code?
thanks

Jaeyoung-Lim, 2020-03-25:
@Robby119 It is usable for hexacopter. As long as you are running PX4, it doesn't matter which vehicle type you are using

Robby119, 2020-03-25:
Just using it directpy or should i change some part inside the code?  Cause
as far as i know, quadcopter and hexacopter have different dynamic model
and it's sure will lead to difference angular acc

Jaeyoung-Lim, 2020-03-25:
@Robby119 We are not using vehicle specific dynamics in this project, but rather using setpoints such as bodyrate / velocity / position which does not matter which airframe you are using

Robby119, 2020-03-25:
Alright, thank you so much for the responses

Robby119, 2020-06-05:
Hi, there's another issue that make me confuse, in this section you're clearly using Faesler differential flatness 
 
[image],

`but when i look at that paper, the equation he's using is different from what you implement in this section.
[image]

It's i misunderstood or what
thanks

Jaeyoung-Lim, 2020-06-05:
@Robby119 I think that looks more or less correct, but the details might be a little different. My intention was not to implement the paper, but have a stable trajectory tracking controller.

closed 2020-10-12 by Jaeyoung-Lim