Jaeyoung-Lim/mavros_controllers #124: implementing mav_trajectory_generation

luisp23, 2020-06-03:
Hello, I want to implement the [ethz-asl/mav_trajectory_generation](https://github.com/ethz-asl/mav_trajectory_generation) to generate trajectories that can be used by the geometric controller. From my understanding of the example code, given a goal position and velocity it will generate coefficients for a polynomial trajectory of message type [PolynomialSegment.msg](https://github.com/ethz-asl/mav_comm/blob/master/mav_planning_msgs/msg/PolynomialSegment.msg). I believe the coefficients x, y, z are for position coefficients and rx, ry, rz are for velocity coefficients. Is this correct? If so, in your [polynomialtrajectory.cpp](https://github.com/Jaeyoung-Lim/mavros_controllers/blob/master/trajectory_publisher/src/polynomialtrajectory.cpp) would these be the coefficients i set` c_x_, c_y_ , c_z_ `to be in the `generatePrimitives(Eigen::Vector3d pos, Eigen::Vector3d vel)` method? 

Also in [trajectoryPublisher.cpp](https://github.com/Jaeyoung-Lim/mavros_controllers/blob/master/trajectory_publisher/src/trajectoryPublisher.cpp) you call updatePrimitives(), you pass `p_mav_` and `v_mav_` which are used to set the coefficients (`c_x_, ... `). 
However, `p_mav_` and `v_mav_` are set in the` mavposeCallback` and `mavtwistCallback `subscribing to `/mavros/local_position/pose` and `/mavros/local_position/velocity` . I thought these topics gave you the current pose and twist estimates from PX4 on the quadcopter is this correct? Why do you set your coefficients to these values? 

These are the snippets of code for reference: 
in trajectoryPublisher.cpp
```
mavposeSub_ = nh_.subscribe("/mavros/local_position/pose", 1, &trajectoryPublisher::mavposeCallback, this,ros::TransportHints().tcpNoDelay());
mavtwistSub_ = nh_.subscribe("/mavros/local_position/velocity", 1, &trajectoryPublisher::mavtwistCallback, this,ros::TransportHints().tcpNoDelay());

void trajectoryPublisher::updatePrimitives(){
  for(int i = 0; i < motionPrimitives_.size() ; i++ ) motionPrimitives_.at(i)->generatePrimitives(p_mav_, v_mav_);
}

void trajectoryPublisher::mavposeCallback(const geometry_msgs::PoseStamped& msg){
  p_mav_(0) = msg.pose.position.x;
  p_mav_(1) = msg.pose.position.y;
  p_mav_(2) = msg.pose.position.z;
  updatePrimitives();
}

void trajectoryPublisher::mavtwistCallback(const geometry_msgs::TwistStamped& msg) {
  v_mav_(0) = msg.twist.linear.x;
  v_mav_(1) = msg.twist.linear.y;
  v_mav_(2) = msg.twist.linear.z;
  updatePrimitives();
}
```
in polynomialtrajectory.cpp

```
void polynomialtrajectory::generatePrimitives(Eigen::Vector3d pos, Eigen::Vector3d vel){
  //Generate primitives based on current state for smooth trajectory
  c_x_(0) = pos(0);
  c_y_(0) = pos(1);
  c_z_(0) = pos(2);

  c_x_(1) = vel(0);
  c_y_(1) = vel(1);
  c_z_(1) = vel(2);
}
```

Thank you again and any clarity/help on these questions would be greatly appreciated!

Jaeyoung-Lim, 2020-06-03:
@luisp23 I don't think generate coefficients are relevant for your use case. 
- What you need is to add a implementation where the trajectory publisher receives the trajectory message from mav_trajectory_generation. What kind of message are you sending your trajectories out? Geometric_controller already has such interface, but only takes the first reference as a input. https://github.com/Jaeyoung-Lim/mavros_controllers/blob/master/geometric_controller/src/geometric_controller.cpp#L134 You can extend this for your use case.
- mavros also has a trajectory_setpoint plugin which directly consumes the trajectory messages

luisp23, 2020-06-03:
`mav_trajectory_generation` outputs a message of type [mav_planning_msgs/msg/PolynomialTrajectory.msg](https://github.com/ethz-asl/mav_comm/blob/master/mav_planning_msgs/msg/PolynomialTrajectory.msg)
So you're saying this already a polynomial trajectory of positions and velocities and not coefficients for one? So then all I need to do is convert these to a` trajectory_msgs::MultiDOFJointTrajectory`   to use it with geometric controller? 

I will take a look at the trajectory_setpoint plugin thank you.

Jaeyoung-Lim, 2020-06-03:
@luisp23 Ah, in that case you need to convert the mav_planning_msgs/msg/PolynomialTrajectory.msg to the MultiDOFJointTrajectory message

luisp23, 2020-06-03:
Okay I understand this a lot better now. Thank you!

closed 2020-10-12 by Jaeyoung-Lim