Jaeyoung-Lim/mavros_controllers #125: Build error related to FlatTarget msg

mzahana, 2020-06-05:
Hi @Jaeyoung-Lim 
Following the build instructions in the README, I get the following error after `catkin build`
```sh
 fatal error: controller_msgs/FlatTarget.h: No such file or directory
 #include <controller_msgs/FlatTarget.h>
```
Do I need to modify the msg dependencies in the geometric_controllers pkg?

mzahana, 2020-06-05:
Solved by changing the config of the `catkin_ws` as follows
```sh
catkin config --merge-devel
```

closed 2020-06-05 by mzahana

Jaeyoung-Lim, 2020-06-05:
@mzahana would you mind adding it to the build instructions in the readme?

mzahana, 2020-06-05:
@Jaeyoung-Lim  here you go https://github.com/Jaeyoung-Lim/mavros_controllers/pull/126

tejalbarnwal, 2023-08-22:
Hey @mzahana, 
How did you figure out to use the `merged` layout?

Jaeyoung-Lim, 2023-08-22:
@tejalbarnwal 
```
        catkin config --merge-devel
```

tejalbarnwal, 2023-08-22:
Hey @Jaeyoung-Lim ,
First of all, thank you so much for creating this excellent repo!
Next, coming to this thread, pardon me if I came across as little confusing with my choice of words, but I wanted to know why does doing `catkin config --merge-devel` work. Doing `catkin config --merge-devel` creates a merged layout as mentioned [here](https://wiki.hanzheteng.com/development/cmake/catkin-tools#catkin-devel-space), leading to `catkin build` working perfectly without giving out the error message related to FlatTarget msg.