Jaeyoung-Lim/mavros_controllers #127: Unstable behaviour when setting velocity_yaw to true

mzahana, 2020-06-22:
Hi @Jaeyoung-Lim 

I would like the drone's heading to be in the velocity direction, but I am not sure if the controller is properly implemented for that. So, I set [velocity_yaw](https://github.com/Jaeyoung-Lim/mavros_controllers/blob/master/geometric_controller/src/geometric_controller.cpp#L40) to true. Any idea what could be the issue?

Jaeyoung-Lim, 2020-06-22:
@mzahana A log would be helpful. Are you running the latest master?

Jaeyoung-Lim, 2020-06-22:
Thanks for reporting, I can reproduce the issue. Added a fix in #128

closed 2020-06-22 by Jaeyoung-Lim

Jaeyoung-Lim, 2020-06-23:
@mzahana Can you test it out and verify that this has been fixed?

mzahana, 2020-06-23:
@Jaeyoung-Lim So you basically removed the multiplication by -1 [here](https://github.com/Jaeyoung-Lim/mavros_controllers/blob/master/geometric_controller/src/geometric_controller.cpp#L418) ?

I tested that, and the takeoff is weird, afterwards it stabilizes. But such unstable takeoff is dangerous on real drones.

See the below video for a test.

[test.zip](https://github.com/Jaeyoung-Lim/mavros_controllers/files/4818493/test.zip)

Jaeyoung-Lim, 2020-06-23:
@mzahana I believe the unstable takeoff is not related to this issue. Could you maybe retune the position controller so that the takeoff is less aggressive?

mzahana, 2020-06-23:
@Jaeyoung-Lim Which position controller are you referring to?

Jaeyoung-Lim, 2020-06-23:
@mzahana The `geometric_controller` in this repo. I also see the aggressive takeoff and seems like the takeoff procedure has changed in the firmware side.

mzahana, 2020-06-23:
@Jaeyoung-Lim I am confused, I thought attitude rates and thrust setpoints are used. Therefore, the PX4 position controller is not used. Is there a takeoff procedure in the `geometric_controller` that uses the PX4 position controller?

Jaeyoung-Lim, 2020-06-23:
@mzahana No, you are right, it uses attitude rates and thrust setpoints.

The `geometric_controller` IS a position controller. I meant retuning the PID gains inside the `geometric_controller` which defines the reference acceleration.

mzahana, 2020-06-23:
I still don't understand how would PX4 differentiate a takeoff maneuver from another arbitrary maneuver when it receives attitude rates as OFFBOARD commands. I believe that it would just track them, whether they correspond to takeoff or not. What do you think?

For the geometric_controller gains, you mean [those](https://github.com/Jaeyoung-Lim/mavros_controllers/blob/master/geometric_controller/src/geometric_controller.cpp#L46-L54) ?

Jaeyoung-Lim, 2020-06-23:
@mzahana You might be right, I was just speculating. Yes, that is correct. Usually `Kp_z`, and `Kv_z` will be the relevant ones

shubham-shahh, 2022-03-30:
Hi @Jaeyoung-Lim , is this fixed? because I am facing the same issue

Jaeyoung-Lim, 2022-03-30:
@shubham-shahh It should be fixed.