Jaeyoung-Lim/mavros_controllers #129: Fundamental behind this implementation

kashishdhal, 2020-06-23:
Hello @Jaeyoung-Lim,

I was just wondering that how your code will behave as compared to the one implemented in https://github.com/ethz-asl/mav_control_rw.

The objective is to track a trajectory.

Regards,
Kashish Dhal

Jaeyoung-Lim, 2020-06-23:
What exactly would you like to know?

kashishdhal, 2020-06-23:
I was just exploring different available packages for trajectory tracking. It will be great if you can explain that how your package differs from the other one I pointed out. I mean the pros and cons? How should I decide which one should I go with?

Jaeyoung-Lim, 2020-06-23:
@Jaeyoung-Lim This package is very dependent to how PX4 is implemented, while the other I believe is not the case. Also the other package uses MPC while this uses a geometric controller, so I would say they are very different packages.

kashishdhal, 2020-06-23:
So your package require me to provide the dynamics of the quad before implementing? How critical will the approximating the dynamics would be?

Can we use this controller with both the VICON and GPS?

Jaeyoung-Lim, 2020-06-23:
@kashishdhal No, you do not need to provide the dynamics of the quad. 

VICON and GPS is not related to this package, since the estimation is done by the PX4 Firmware, but yes you can use either if you configure the EKF2 accordingly

kashishdhal, 2020-06-23:
Our EKF2 is already configured. So we just need to provide the trajectory to track?

Jaeyoung-Lim, 2020-06-23:
@kashishdhal I think you will have a better answer to that question, since I am not aware what the status of your system is

kashishdhal, 2020-06-23:
okay thanks!

closed 2020-06-23 by kashishdhal

Jaeyoung-Lim, 2020-06-23:
@kashishdhal Please feel free to post another issue if you have any more questions :smile:

kashishdhal, 2020-06-23:
Sure I am just trying to run you package. I just resolved first error:

<pre><font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~/catkin_ws</b></font>$ rosdep install --from-paths src --ignore-src -y --rosdistro $ROS_DISTRO
<b>executing command [sudo -H apt-get install -y ros-melodic-rosbridge-library]</b>
[sudo] password for gnclab: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package ros-melodic-rosbridge-library
ERROR: the following rosdeps failed to install
  apt: command [sudo -H apt-get install -y ros-melodic-rosbridge-library] failed</pre>

Jaeyoung-Lim, 2020-06-23:
@kashishdhal Do you have ROS installed? which ros distro are you on? 

Since this issue no longer is relevant can you move this discussion to a seprarate repo?

kashishdhal, 2020-06-23:
So I was able to run sitl_trajectory_track_circle.launch successfully. I had to install ros-melodic-rosbridge-library manually. 

However, for the mav_trajectory_track_circle.launch where do I need to provide the trajectory? I mean the type of trajectory, radius of the circle etc? I am using VICON for localization and EKF2 is all set, I can see the output on local_position/pose.

If you do not prefer to answer here, I will create another issue.

Jaeyoung-Lim, 2020-06-23:
@kashishdhal Please post another issue.

You can either take `trajectory_publisher` as an example and pass on reference to `geometric_controller`, or extend the trajectory interface of the `geometric_controller` to follow a trajectory