Jaeyoung-Lim/mavros_controllers #132: Topic names relative to node's namespace

mzahana, 2020-06-28:
Hi @Jaeyoung-Lim 

I noticed that [some subscribers in geometric_controller.cpp](https://github.com/Jaeyoung-Lim/mavros_controllers/blob/master/geometric_controller/src/geometric_controller.cpp#L20-L23) has topics with respect to `/` namespace. Would it be better to keep it with respect to the node's namespace? Just like other subscribers in the same node.

Jaeyoung-Lim, 2020-06-28:
@mzahana That is a very good point. It would make sense that we fix this properly.

The reason I was using absolute namespaces was because usually ROS topic namespaces become quite fragile depending on your launchfile. Therefore I took a easy cut to get it "always" work with mavros. However, this is clearly a hack and would not scale if we have multiple instances.

Do you have any good usecases for this? Did this aspect of the code block you in any usecase?

mzahana, 2020-06-29:
An obvious use case is when you run multiple instances for multi-drone setup. With the current setup it fails. A straight forward fix is to keep the topics and params relative to the node's namespace. Namespaces are expected to be defined in launch files using `<group ns="namepsace">` tags.