Jaeyoung-Lim/mavros_controllers #134: Can I use it in the Rover?

skyaid, 2020-07-07:
Hello

I'm studying the px4 Rover software

I want to use the mavros_controller in the Rover, but the motor rotates fast when converting offboard mode.

Forward servo turns right when converting offboard

Can't I use it in the Rover?

Jaeyoung-Lim, 2020-07-07:
What are you trying to do with the rover? It depends on what setpoints you are sending

lgh5054, 2020-07-08:
@Jaeyoung-Lim https://github.com/mzahana/px4_fast_planner
I'd like to test this.

Jaeyoung-Lim, 2020-07-08:
@lgh5054 In that case no.

The reason is that the body rate controller commands body rates and thrust assuming that it is a underactuated multirotor system (e.g. thrust is computed to stay in hover bodyrates make the vehicle to tilt forward to move forward).

You can maybe try using the trajectory_publisher that publishes a variety of setpoints depending on what you need

closed 2020-07-14 by skyaid