Jaeyoung-Lim/mavros_controllers #142: Drone yaw rotational speed change

lgh5054, 2020-07-18:
It's testing a real drone. The thrust was adjusted low through a geometric structure.
Moving back and forth is slow. But only the yaw rotates very fast. Can yaw rotational speed be changed?

Jaeyoung-Lim, 2020-07-18:
@lgh5054 of course, You need to tune your vehicle properly BEFORE running this ros node on your vehicle.

If you are curious on how to tune your vehicle in px4, please refer to this document: https://docs.px4.io/master/en/config_mc/pid_tuning_guide_multicopter.html

lgh5054, 2020-07-18:
This is not a controller issue. The settings for px4 are complete and there is no problem.

there is a way to reduce the yaw rotation speed of the geometric.cpp? The quick rotation of the yaw can also be seen in the gazebo simulator.

lgh5054, 2020-07-18:
> @ Jaeyoung-Lim 컨트롤러 문제가 아닙니다. px4의 설정이 완료되었으며 아무런 문제가 없습니다.
> 
> geometric.cpp의 요 회전 속도를 줄이는 방법이 있습니까? 요의 빠른 회전은 전망대 시뮬레이터에서도 볼 수 있습니다.

Is this a problem with fc?

Jaeyoung-Lim, 2020-07-18:
@lgh5054 It is not clear what "value" or behavior you are trying to change. Are you trying to disable velocity heading?

lgh5054, 2020-07-19:
@Jaeyoung-Lim thanks I'll be right back with the contents ready

lgh5054, 2020-07-19:
@Jaeyoung-Lim I'd like to slow down the rotation shown in this video. Is it a pid problem?

lgh5054, 2020-07-19:
@Jaeyoung-Lim https://drive.google.com/file/d/1NQvREkDNFnoNQJ_xbVhjA_jyTtF8QdCN/view?usp=drivesdk

I want to reduce the rotation speed of the video. Or is this a pid problem?

Jaeyoung-Lim, 2020-07-19:
@lgh5054 which video?

 The yaw is controlled to track velocity or keep a specific heading. I do t understand what you want to "slow down"

lgh5054, 2020-07-19:
@Jaeyoung-Lim Doesn't the video above play?
the rotational speed of a drone's torso.

Jaeyoung-Lim, 2020-07-19:
@lgh5054 I dont have access to the video...

lgh5054, 2020-07-19:
@Jaeyoung-Lim https://photos.app.goo.gl/6XuRbdgnKaDSYAV36
Can I see the video?😭

lgh5054, 2020-07-23:
I'm sorry, I'm closing the question 

This question does not apply here.

closed 2020-07-23 by lgh5054