Jaeyoung-Lim/mavros_controllers #143: What is the reason of setting norm_thrust_const and norm_thrust_offset to 0.05 and 0.1?

rainfall1998, 2020-07-26:
I am trying to get the controller work in Airsim, thus I can use Airsim API 
```
moveByAngleRatesThrottleAsync(roll_rate, pitch_rate, yaw_rate, throttle, cmd_time_)
```
while I find these two value just work perfectly in Airsim. because 0.05 * g_ + 0.1 = 0.59 (throttle), which is just the hover throttle in airsim. So I was wondering if there is any reason to set these two parameter.

Jaeyoung-Lim, 2020-07-26:
@rainfall1998 Yes, because depending on your hardware you would need to change those values according to your system

HenryHuYu, 2020-07-26:
> @rainfall1998 Yes, because depending on your hardware you would need to change those values according to your system

Thanks for you reply. I also have the same question. And does **norm_thrust_const** have any physical meaning? according to my understanding, it is something like "throttle per accelerate"?

Robby119, 2020-08-12:
Hi @Jaeyoung-Lim i'm trying to understand how to calculate thrust in your code

 ratecmd(3) = std::max(0.0, std::min(1.0, norm_thrust_const_ * ref_acc.dot(zb) + norm_thrust_offset_));  // Calculate thrust

do you have paper/report that can help me to understand how it works ?

Thanks in advance

Jaeyoung-Lim, 2021-03-13:
@Robby119 PX4 simply uses a normalized thrust (-1 ~ 1), and I am just trying to convert my metric thrust inputs into normalized thrust inputs

closed 2021-03-13 by Jaeyoung-Lim