Jaeyoung-Lim/mavros_controllers #144: Controller tuning.

HenryHuYu, 2020-07-28:
[Drone2020v1   2020-07-28 15-27-23.zip](https://github.com/Jaeyoung-Lim/mavros_controllers/files/4989345/Drone2020v1.2020-07-28.15-27-23.zip)
Hi, I am trying to tune the controller in Airsim. For the video it is the same as the parameter as the original code. The video shows that the is kinda viberating.  To my understanding, it is because the **Kp** is too big or the **attitude_tau** is too small, but as I make them smaller or make one of them smaller, the drone is still viberating,.
Also, I design a polynomial trajectory for execute time T = 3, and at the time t = 3, v(t)  = 0; a(t) = 0;  For the current parameter, it will go back and forth around the final target position. 
So do you have any advices on how to tune the parameter? and will it help if I use publist flat target witch contains (p, v, a)? currently I publish only **p** and **v**
I am new to tune the controller.  really appreciated. Thanks for your help!

Jaeyoung-Lim, 2021-02-02:
@HenryHuYu Probably for tuning, it would be better if you just publish a stationary position setpoint and move it from there

Kp is a gain for the position controller and attitude_tau is the rate controller time constant. 
If it still oscillates you need to probably make it bigger

mzahana, 2021-09-13:
@Jaeyoung-Lim I am also wondering what would be the best approach to tune this controller for real vehicles. How would you start with initial values based on a given vehicle specs?

Jaeyoung-Lim, 2021-09-13:
@mzahana My recommendation is that I creat a model that is as similar as the real vehicle ( including px4 tuning gains) and tune this controller on top of that model