Jaeyoung-Lim/mavros_controllers #145: Add Waypoint Function

lgh5054, 2020-07-29:
Hello
Thank you always for your help.
Only offboard mode seems to work on the Mavros_controller. Can't I upload the mission through QGC?

HenryHuYu, 2020-07-30:
> Hello
> Thank you always for your help.
> Only offboard mode seems to work on the Mavros_controller. Can't I upload the mission through QGC?

I believe this controller mainly focus on the autonomous drone flying and with sensors and computations only onboard. And  to my understanding, mission mode in QGC is something more like using "mavros setpoint position", the controller here is more flexible and can do more aggressive things.

Jaeyoung-Lim, 2020-09-18:
@lgh5054 PX4 already supports waypoint missions. Why would you want to upload a mission and run it in offboard mode?

lgh5054, 2020-09-18:
@Jaeyoung-Lim We would like to implement the currently implemented path planning algorithm like a px4 dedicated local_planner

Jaeyoung-Lim, 2020-09-18:
@lgh5054 Okay, then you need to implement yourself how to handle the waypoints

lgh5054, 2020-09-18:
@Jaeyoung-Lim Thank you for your answer. Close the comment

closed 2020-09-18 by lgh5054