Jaeyoung-Lim/mavros_controllers #148: How to Match Commanded Acceleration with the measured acceleration

kashishdhal, 2020-08-26:
Hello @Jaeyoung-Lim ,

In continuation to #130 

My angular velocity plots for one of the flights (omega = 2 rad/s) are here:

[image]
[image]

I am guessing the tuning is good on PX4 side?  But please note that these plots are not for the same flight that I showed you earlier for acceleration. I am sharing this because it shows good tracking of the angular velocities.

Log File Link is here: https://logs.px4.io/plot_app?log=ca8f0cea-0534-4319-8e99-71fee6b91cf0

kashishdhal, 2020-08-26:
Hello @Jaeyoung-Lim ,

If you cannot think of anything that is fine, but just let me know.

Jaeyoung-Lim, 2020-08-26:
@kashishdhal As you can see in the logs, the rate tracking performance is different on roll and pitch

kashishdhal, 2020-08-26:
@Jaeyoung-Lim , upon looking at the logs, I can't really think of the difference between the two, I can see that they both are being tracked pretty nicely as per the commanded rates.

Anyways, would you be able to explain a bit what difference do you see in the two? Since I am having offset in x-axis so I am guessing there is some problem in pitch-rate?

Jaeyoung-Lim, 2020-08-26:
@kashishdhal I cannot tell you what is going on, since this involves your tuning and your hardware. What you are showing me is only the response of the system.

 but clearly you can see the difference of tracking performance in rate tracking in the logs and I suspect this brings you the acceleration tracking difference

kashishdhal, 2020-08-26:
@Jaeyoung-Lim , thanks for the reply.

I was asking you what difference do you see in the performance, such as overshoot/undershoot, delay, noise, etc. It will be helpful if you can define the performance difference in engineering terms so that I can go back and tune my system and then see if that particular problem is resolved.

Sorry I might be asking a stupid question but I don't see what you are referring to!

kashishdhal, 2020-08-26:
You can refer to these two pics for our discussion:

[image]
[image]

Jaeyoung-Lim, 2020-08-26:
@kashishdhal Unfortunately the data you have shown is not representative to make those statements. You need a controlled environment to make those statements e.g. giving step inputs. 

If you are unsure, there are tons of materials on how you can design your input to make your response representative in the system identification/adaptive control literature. So please look it up.

 I unfortunately dont have time to analyze your logs, but on hindsight looks like the roll and pitch tracking performance is not the same (look at the setpoint vs estimated and how close the two lines are), and that was what I was referring to.

kashishdhal, 2020-08-26:
Well, that is correct, but we can still see overshoots/undershoots etc in the response, for example, if I will increase the value of P gain there there will be more overshoot/ less delay even in the oscillating signals. I was referring to that. 

But, thanks for the reply, I got an idea, I can understand you that have don't have enough time to figure out everything. We really appreciate  your time and the your contribution to the community.

One of the reason of un-even tracking in the roll and pitch axis could be the weight distribution (the battery is rectangular instead of square). However, we will try to tune our system to make it better track in both axis. Thanks again!

Jaeyoung-Lim, 2020-08-26:
No problem! Please share on how it goes, and if you can come up with tooling to make trouble shooting easier please contribute back!

> Well, that is correct, but we can still see overshoots/undershoots etc in the response, for example, if I will increase the value of P gain there there will be more overshoot/ less delay even in the oscillating signals. I was referring to that.

Yes, but for me it is very hard to make any assessment since there is too much happening. You need to isolate the problem to be able to make those assessments. Otherwise, it is highly probable that the assessments are wrong, and it doesn't make sense to test fly the vehicle based on the assumptions.

kashishdhal, 2020-08-27:
Yes, I can share how I tuned my PX4 attitude rate PID loop. If you remember, you asked me to fly the drone in acro mode for tuning. But I found out that it was not a good method and instead I gave step input commands of roll-rate, pitch-rate independently one by one to tune. So basically one can fly in Acro from Offboard control to tune which is again used for geometric controller.

HenryHuYu, 2020-09-20:
> Yes, I can share how I tuned my PX4 attitude rate PID loop. If you remember, you asked me to fly the drone in acro mode for tuning. But I found out that it was not a good method and instead I gave step input commands of roll-rate, pitch-rate independently one by one to tune. So basically one can fly in Acro from Offboard control to tune which is again used for geometric controller.
> 
> I will share the other methods I will employ to make tuning better or troubleshooting. Thanks

Hi! Could you please give a hint how to "gave step input commands" with this word and tune the controller?

kashishdhal, 2020-09-20:
@HenryHuYu , you can give the step input by writing C++ file in the following steps:
1. Use the position controller to take the drone to a certain height
2. Switch to the rate controller and give rate command one by one in each axis, such as 20 rad/s for 3 sec then give -20 rad/s for 3 sec so that the drone does not crash. If you don't know how to send rate commands, take mavros_controller as an example.
3. Stop sending the setpoints which will make the drone land.

kashishdhal, 2020-09-20:
@Jaeyoung-Lim, so I was able to match the acceleration commands to a reasonable value. I purchased two batteries, connected them to make a square like shape so that the weight distribution is even. Then, tuned my PX4 side attitude rate controller again. 
Here is the tracking for omega = 2 rad/s, radius = 0.75 m
[image]

However, I would like to make a few recommendations to you:
1. You are continuously publishing a circular trajectory, during the take-off, error in position and velocity are large which makes acceleration commands too high, that was the reason for a lot of crashes in the beginning. At the initial stage, I had to modify your package to make it work, I changed the lower limit of Kp_x, Kp_y, Kv_x, Kv_y to 0.1 because in your package the lower limit is set to 1. Later when drone is in the air I quickly change them to 3,3,1,1. I know this is not a good fix, ideally to fix this issue we should generate a smooth trajectory which will take us to the circle then switch to the circular trajectory. I guess that is how Mellinger did.
2. The second way, which I am using right now, is to use position controller to take the drone to a certain height and then switch to the circular trajectory. It works for me, no crashes even for higher values of omega. 

I am sharing my experience since you asked to give feedback but it will be upto you to make those changes or leave the package as is. I really appreciate your contribution to the community. Thanks!

closed 2020-09-20 by kashishdhal

Jaeyoung-Lim, 2020-09-20:
@kashishdhal Thanks for the feedback and sharing the results!

Is there any chance that you can contribute the changes you mentioned to improve this package? 

Thanks,

kashishdhal, 2020-09-20:
@Jaeyoung-Lim , yes, I can help! How should we do that?

Jaeyoung-Lim, 2020-09-20:
@kashishdhal You can make a pull request with the modifications towards this repository.

If you are not sure how to do this, [here](https://docs.github.com/en/github/collaborating-with-issues-and-pull-requests/creating-a-pull-request) is an instruction how. This repository has benefited greatly from various contributions, and I would appreciate if you can contribute too.

kashishdhal, 2020-09-20:
@Jaeyoung-Lim , I have used acceleration set-point in PX4, I do not know if it will be the same or they are making heavy changes. Also, I have used the geometric controller to send the acceleration set-points. 

The problem with acceleration set-point is that you will end-up having drift in the position and velocity. The nice thing about using geometric controller is that since you also provide reference position and velocity, it will kill the drift. From my experience, acceleration set-points are a little tricky as compared to the velocity or position set-points.

I will go through the article and create a pull request.

kashishdhal, 2020-09-21:
@Jaeyoung-Lim , should I add the position controller to take the drone to a certain altitude using a separate package? We can call it position controller and I can put under the geometric controller? Is that fine?