Jaeyoung-Lim/mavros_controllers #149: How to correctly publish a trajectory to be followed by the geometrical controller

maxhofidrone, 2020-09-11:
Hi @Jaeyoung-Lim ,
I have some troubles to correctly understand how to publish a trajectory to the geometrical controller which will then itself command the /mavros/setpoints to the FC.
Up to now I'm using your provided example launched with  sitl_trajectory_track_circle.launch.
According your information provided to me via Slack, the geometrical controller receive the next  trajectory setpoint to go here:
https://github.com/Jaeyoung-Lim/mavros_controllers/blob/master/geometric_controller/src/geometric_controller.cpp#L22

But I can not see that topic when looking on the available topics running the mentioned example:

[image]

I also don't understand the /reference/ namespace topics. Are they used for the visualization with RViz?

I was planning my first "trajectory" generator to provide some step-response trajectory to the geometrical controller e.g publishing a MultiDOFJointTrajectory message of Point A(0,0,2) for 15 seconds with 100Hz, then publishing the new setpoint B(0,0,3) for another 15 seconds. My exceptions would be that the geometrical controller aims to hold the drone as accurate as possible at point A, after the step at point B. In a next step I would probably try to publish some circular trajectory like in your example or a parabolic shaped trajectory.

Thanks for any help and clarification in advance!
Max

RapIt, 2020-09-16:
Did you make any progress here? I have exactly the same scenario, I just want the UAV to follow a path of points using the position controller. A comprehensive example of publishing some setpoints with an associated time span would be so great, if someone already did that. I tried to publish messages to the `/command/trajectory` topic but apparently this doesn't have an effect while running the circle example.

maxhofidrone, 2020-09-18:
Not yet. I think you can not publish different setpoints (yours and the circle ones) at the same time. You probably should replace the circle trajectory messages with your desired path.
But still I don't understand the correct way to publish a trajectory to the geometrical controller based on the instructions and the circle example.

Jaeyoung-Lim, 2020-09-18:
@maxhofidrone You need to "replace" the trajectory_publisher to publish trajectories to the geometric_controller.

trajectory_publisher is an example of an external application sending external references to the geometric_controller

HenryHuYu, 2020-09-20:
> Not yet. I think you can not publish different setpoints (yours and the circle ones) at the same time. You probably should replace the circle trajectory messages with your desired path.
> But still I [...]

I just made it work in AirSim, it is a great work and thanks to the author again. First of all you need to understand path and trajectory, to my understanding, trajectory is a function of time, (like the circle trajectory offered by the author.) and then you need to sample it. Sampling mean that you need to sample out position, velocity, accelerate at certain rate. (again like the circle trajectory, you sample the trajctory (p,v,a) at time 0.01, and the (p, v, a) at time 0.02 .... this sample result is the reference target of the geometric controller)

Jaeyoung-Lim, 2020-09-20:
@HenryHuYu Awesome! Are there any videos? I would be very interested in seeing a video if you can share! Thanks!

HenryHuYu, 2020-09-20:
> @HenryHuYu Awesome! Are there any videos? I would be very interested in seeing a video if you can share! Thanks!

Yes, sure. I would like to share a competition video that we join. Since the AirSim uses NED coordinate system, I made some change on the code (like the g is (0, 0, 9.8) instead of (0, 0, -9.8)), but I realize that the modification might be unnecessary. I could do another coordinate transform instead. 

The Video link is as follows: 
[https://www.bilibili.com/video/BV1UK411P7Et/](url)

We modified some parameters and set the max_vel of trajectory very low (1m/s) so that we can track it well. However there are lots of problem we met when tuning the controller. These are my main concerns:
1. I noticed that in your launch file, tracking a circle trajectory and tracking a polynomial trajectory have different Kp values. What is the main reason for this, why does the trajectory (not only hardware itself) affect the parameters? 
2. Is there a way to combine yaw trajectory into a position trajectory, or yaw is always set seperately? 

Thank you again!

closed 2021-03-13 by Jaeyoung-Lim