Jaeyoung-Lim/mavros_controllers #150: Attitude error different from the referenced paper

quarcle, 2020-09-11:
Why does the attitude controller in the [geometric_attcontroller](https://github.com/Jaeyoung-Lim/mavros_controllers/blob/master/geometric_controller/src/geometric_controller.cpp) different from the mentioned paper "Geometric tracking control
  // of a quadrotor UAV on SE (3)" by Lee, et al?

I tried to change the attitude error to be `error_att = 0.5 * matrix_hat_inv(rotmat_d.transpose() * rotmat - rotmat.transpose() * rotmat_d);` to make it similar to the paper, but it turns out that the quadcopter falls down in the Gazebo simulator. Do you know why that happens?

Many thanks.

Jaeyoung-Lim, 2020-09-22:
@quarcle Can you share the log?

quarcle, 2020-09-22:
Here is the log file:
[roslaunch-qeui-X71-6589.log](https://github.com/Jaeyoung-Lim/mavros_controllers/files/5261726/roslaunch-qeui-X71-6589.log)

Might that help?

There is no error shown while executing the controller, but the drone crashes after taking off and trying to follow a specified trajectory (in my case, I run `sitl_trajectory_track_circle.launch`)

Jaeyoung-Lim, 2020-09-22:
@quarcle I meant the flight log. I do see that the equation is wrong, but not sure why it flies now

quarcle, 2020-09-22:
Which log do you mean? I attach two logs (roslaunch log and the ulog files) as follows:
[geocontrol_logs.zip](https://github.com/Jaeyoung-Lim/mavros_controllers/files/5261966/geocontrol_logs.zip)

quarcle, 2020-09-30:
> error_att = 0.5 * matrix_hat_inv(-rotmat_d.transpose() * rotmat + rotmat.transpose()

What do you mean?

quarcle, 2020-09-30:
> error_att = 0.5 * matrix_hat_inv(-rotmat_d.transpose() * rotmat + rotmat.transpose()

@valenbase what does it mean?

valenbase, 2020-10-10:
> > error_att = 0.5 * matrix_hat_inv(-rotmat_d.transpose() * rotmat + rotmat.transpose()
> 
> @valenbase what does it mean?

i mean exchange content on both sides of the symbol -

closed 2021-02-02 by Jaeyoung-Lim

quarcle, 2021-02-02:
Has anyone found out the problem and how to fix it?

Jaeyoung-Lim, 2021-02-02:
@quarcle Could you clarify what the problem is?

quarcle, 2021-02-02:
[This code](https://github.com/Jaeyoung-Lim/mavros_controllers/blob/master/geometric_controller/src/geometric_controller.cpp) line 444 uses `error_att = 0.5 * matrix_hat_inv(rotmat_d.transpose() * rotmat - rotmat.transpose() * rotmat);` while the referenced paper uses `error_att = 0.5 * matrix_hat_inv(rotmat_d.transpose() * rotmat - rotmat.transpose() * rotmat_d);`. When I changed to the error equation mentioned in the paper, the copter crashes. How should I resolve the problem?