Jaeyoung-Lim/mavros_controllers #151: PX4 controllers comparison

08beeqtufail-sudo, 2020-09-11:
Hello @Jaeyoung-Lim ,
First of all, thank you for the sharing the work you have done. I was able to run your controllers with PX4-Gazebo simulation but that raised a question in my mind:
Since we already have position, attitude and rate controllers implemented in px4, where does your controller fit in the px4 architecture? Does it bypass the px4's low level controllers? If not, then how do both controllers co-exist? Thanks in advance for your answer as it will help me clarify some doubts. 
Regards

Jaeyoung-Lim, 2020-09-21:
@08beeqtufail-sudo Sorry for the late reply, this package is utilizing [**OFFBOARD Mode**](https://docs.px4.io/master/en/flight_modes/offboard.html) which allows you to forward setpoints such as position setpoints, velocity setpoints, attitude setpoints, angular rate setpoints to PX4.

For example, `geometric_controller` is sending body rate setpoints to the mc_att_controller module inside PX4. The `trajectory_publisher` can send position setpoints with velocity feedforward setpoints to the `mc_pos_controller` module inside PX4.

Does this answer your question?

closed 2021-03-13 by Jaeyoung-Lim

fangjir, 2023-09-07:
hello,@Jaeyoung-Lim 
if i only is sending body rate setpoints to the px4,without sending position setpoints and velocity feedforward setpoints to the mc_pos_controller ?then whether px4 can work well?

Jaeyoung-Lim, 2023-09-07:
@fangjir Yes, that is what this repo was built for