Jaeyoung-Lim/mavros_controllers #153: Including Yaw setpoint in the FlatTarget Msg

mzahana, 2020-10-05:
Hi @Jaeyoung-Lim 

I was looking at the [flattargetCallback](https://github.com/Jaeyoung-Lim/mavros_controllers/blob/master/geometric_controller/src/geometric_controller.cpp#L101-L133), and I don't see it accepting yaw setpoint. I know there is another callback just for yaw setpoint [yawtargetCallback](https://github.com/Jaeyoung-Lim/mavros_controllers/blob/master/geometric_controller/src/geometric_controller.cpp#L135). I was wondering why not to also add yaw setpoint in the [FlatTarget.msg](https://github.com/Jaeyoung-Lim/mavros_controllers/blob/master/controller_msgs/msg/FlatTarget.msg) in order to allow all setpoints to be set in one message and published to a single topic with the same timestamp instead of having them separated in two different topics which could possibly be published at different timestamps (even for the same setpoints). Is there a reason for that?

Thanks.

Jaeyoung-Lim, 2020-10-05:
@mzahana You are right, this was something that was at the back of my head, but didn't have time to do.

One note is that we need a way to invalidate yaw setpoint in case you want to ignore it

FaboNo, 2020-10-09:
@Jaeyoung-Lim Following @mzahana if we consider that the drone is equipped with a camera, it will be great to include the yaw and yaw velocity in the flattargetCallback and thus in the computeBodyRateCmd. What will be the modification in the equations and if we want to implement it, which paper(s) do we have to read? The idea is to couple fastplanner (from hkust) and mavros_controller.

mzahana, 2020-10-09:
@FaboNo Have you looked at [px4_fast_planner](https://github.com/mzahana/px4_fast_planner) package where I did the required interface between FastPlanner and mavros_controllers?

mzahana, 2020-10-09:
The missing piece is the yaw rate which is not implemented in the geometric_controller, but it's doable.

Jaeyoung-Lim, 2020-10-09:
@FaboNo Nothing much to add since yaw is already handled in the controller through a different message. 

As @mzahana mentioned, he already integrated into the HKUST fast planner in https://github.com/mzahana/px4_fast_planner so might be worth looking into that.

On adding yaw rate, I am not sure if is worth it. Quadrotor systems are differentially flat and yaw rate is usually not part of the flat outputs - meaning that yaw rate commands are unnecessary in most cases

FaboNo, 2020-10-12:
@mzahana thank you I will have a look
@Jaeyoung-Lim yes you are right, but I was thinking it may be interesting to limit the rate to avoid slam issues (camera blurring or too large rotation in case of Lidar)

Jaeyoung-Lim, 2020-10-12:
@FaboNo Right :smile: If you are looking into reducing the yaw gain, it might be worth digging this back up: https://github.com/Jaeyoung-Lim/mavros_controllers/pull/35

The intention was to respect vehicle limits, since quadrotors usually suffer from large yaw step inputs but I think this will be also applicable for perception aware control

FaboNo, 2020-10-14:
@Jaeyoung-Lim Ah interesting I will dig into it:)

Jaeyoung-Lim, 2020-10-23:
@FaboNo Any updates? I would be happy to support you on this

FaboNo, 2020-10-25:
@Jaeyoung-Lim 

I read the two articles you mentioned to understand how you implemented the geometric controller and I have few questions:

in `void geometricCtrl::computeBodyRateCmd(bool ctrl_mode)`
`a_rd = R_ref * D_.asDiagonal() * R_ref.transpose() * targetVel_`;  is always equal to 0 since D_ is a null matrix (dx,dy,dz) = (0,0,0)

The geometric controller with mavros/PX4 implement the control laws from Faessler et al. paper but without the rotor drag components because it is easier to integrate with Mavros/px4? Indeed other implementation such as Lee et al requires the knowledge of J (inertia matrix), mass ...

In  `Eigen::Vector4d geometricCtrl::acc2quaternion(Eigen::Vector3d vector_acc, double yaw)` you wrote:
```
if (velocity_yaw_) 
  		proj_xb_des = targetVel_.normalized();
  	else 
  		proj_xb_des << std::cos(yaw), std::sin(yaw), 0.0;

  	zb_des = vector_acc / vector_acc.norm();
  	yb_des = zb_des.cross(proj_xb_des) / (zb_des.cross(proj_xb_des)).norm();   
  	xb_des = yb_des.cross(zb_des) / ( yb_des.cross(zb_des) ).norm();
```
But in the Faessler et al. paper :
xb_des = y_c x z_b_des/ (y_c x z_b_des).norm  with y_c = (-sin(phi), cos(phi), 0) 
I guess it is ok because you took y_b_des instead with x_b_projected.
but in any case `xb_des = yb_des.cross(zb_des)`, i.e. no need to normalized since the two vectors (yb_des and zb_des) are already normalized.

In `Eigen::Vector4d geometricCtrl::attcontroller(Eigen::Vector4d &ref_att, Eigen::Vector3d &ref_acc, Eigen::Vector4d &curr_att)` I do not see from where the following equation comes from:
`ratecmd(3) = std::max(0.0, std::min(1.0, norm_thrust_const_ * ref_acc.dot(zb)))`
what is the role of the `norm_thrust_const_` here?

On another thread, regarding a real implementation on a drone, you mentioned three steps:
a. You need to make sure the rate controller of the px4 side is tuned properly so that it is capable of tracking the angular rate commands you are providing.
b. You need to make sure the normalized thrust inputs on the geometric_controller side is tuned properly
c. Only when a,b is done you need to tune the K_pos and K_vel terms

Can you tell me what do you mean in point (b) ?

I started to look at the pr #35 and look at this link:
https://github.com/Jaeyoung-Lim/mavros_controllers/pull/35/commits/8076712ef8e79a196e3c34c60524da162ad0b443

Does it mean that you implemented the possibility to include a yaw_rate in the controller in another branch?

Thanks again for your support

Jaeyoung-Lim, 2020-10-25:
> in void geometricCtrl::computeBodyRateCmd(bool ctrl_mode)
a_rd = R_ref * D_.asDiagonal() * R_ref.transpose() * targetVel_; is always equal to 0 since D_ is a null matrix (dx,dy,dz) = (0,0,0)

The geometric controller with mavros/PX4 implement the control laws from Faessler et al. paper but without the rotor drag components because it is easier to integrate with Mavros/px4?

FaboNo, 2020-10-26:
@Jaeyoung-Lim great explanations! Yes it helps a lot to clarify the points which were unclear to me... and it confirms that implementing a geometric controller on a real drone is not an easy task!

Jaeyoung-Lim, 2020-10-26:
@FaboNo In the end, it is not that complicated. It is just that it is not easy to control force with normal ESCs since there is no feedback therefore it is better to avoid using force / moment based controls in the first place.

Jaeyoung-Lim, 2021-03-13:
@mzahana Are you still using this package and have plans to move forward with the yaw setpoints?

mzahana, 2021-03-13:
@Jaeyoung-Lim Sorry for getting disconnected from this issue, I am short on time at the time being. I use this package on demand, i.e. depending on the project(s) I work on. At the time being, I am not using it. However, I still believe that the yaw setpoint should be in the flat target callback, and msg. I have had the intention to submit a PR for this, but not soon as I am swamped with other tasks with higher priorities. So, maybe next month, unless someone else is generous and does it earlier.

On a side note, I was also interested to add integral action in the controller to correct for steady-state error as the current controller acts like a PD and won't respond to steady-state errors. I saw one paper from Martin Saska's group did that, and since then I had the intention to do it, but again, short on time! This would be a separate issue/PR though.

Thanks for the follow up.

958117216, 2021-05-23:
@mzahana man，can you share the  paper from Martin Saska's group ? I will appreciate it.

mzahana, 2021-05-26:
@958117216 
The paper is called Autonomous Landing on a Moving Vehicle with an Unmanned Aerial Vehicle", see section 9.

[Link to the paper](https://www.google.com/url?sa=t&source=web&rct=j&url=https://onlinelibrary.wiley.com/doi/am-pdf/10.1002/rob.21858&ved=2ahUKEwiYmOKJlefwAhVThf0HHXPyBc4QFjABegQICRAC&usg=AOvVaw2HxxelRaStqHHtDKzZS6cr)

brunopinto900, 2021-07-19:
I am adding yaw field on the FlatTarget.msg. But i am having a compilation error:

Errors     << geometric_controller:make /home/bruno/catkin_ws/logs/geometric_controller/build.make.004.log                                                
/home/bruno/catkin_ws/src/mavros_controllers/geometric_controller/src/geometric_controller.cpp: In member function ‘void geometricCtrl::flattargetCallback(const FlatTarget&)’:
/home/bruno/catkin_ws/src/mavros_controllers/geometric_controller/src/geometric_controller.cpp:150:35: error: ‘const FlatTarget’ {aka ‘const struct controller_msgs::FlatTarget_<std::allocator<void> >’} has no member named ‘yaw’
  150 |   std_msgs::Float64 yaw_msg = msg.yaw;
      |                                   ^~~

---------------------------------------------------------------------------------------------------------------------------------
This is the message definition:

geometry_msgs/Vector3 position
geometry_msgs/Vector3 velocity
geometry_msgs/Vector3 acceleration
geometry_msgs/Vector3 jerk
geometry_msgs/Vector3 snap
std_msgs/Float64 yaw
------------------------------------------------------------------------------------

This is how i am using it:

void geometricCtrl::flattargetCallback(const controller_msgs::FlatTarget &msg) {
  reference_request_last_ = reference_request_now_;

  targetPos_prev_ = targetPos_;
  targetVel_prev_ = targetVel_;

  reference_request_now_ = ros::Time::now();
  reference_request_dt_ = (reference_request_now_ - reference_request_last_).toSec();

  targetPos_ = toEigen(msg.position);
  targetVel_ = toEigen(msg.velocity);
  std_msgs::Float64 yaw_msg = msg.yaw;
  mavYaw_ = double(yaw_msg.data);

Makes no sense

Jaeyoung-Lim, 2021-07-19:
@brunopinto900 Your message definition is not updated. Do it after 'catkin clean'

brunopinto900, 2021-07-19:
I tried that before opening the issue xD, didn't work either.

Jaeyoung-Lim, 2021-07-19:
@brunopinto900 Then there is something wrong with your workapace path

brunopinto900, 2021-07-20:
Had to execute catkin clean on other workspace, that was somehow interfering with the mavros_controller workspace. Everything is working right now.

brunopinto900, 2021-07-20:
To test the yaw field in the FlatTarget.msg, i requested the quadrotor to fly a circle while pointing (yaw) to the center of rotation. Here is the results:

https://user-images.githubusercontent.com/48806267/126334615-44e0ed2f-14c7-4241-94c3-17b1ac5c9220.mp4

https://user-images.githubusercontent.com/48806267/126334869-498370d2-2b36-4c3a-97b7-25e7df114809.mp4

I added in the shapeTrajectory.cpp the following code:

float shapetrajectory::getYaw(double time) {

  float yaw;

  switch (type_) {
    case TRAJ_ZERO:

      yaw = 0;
      break;

    case TRAJ_CIRCLE:
      yaw = (traj_omega_ * time) - M_PI; // always pointing to the center of the circle
      yaw = atan2f( sin(yaw),cos(yaw) ); // wrap angle around [-pi,pi]
      break;
  }

  return yaw;
}

What do you think?

Jaeyoung-Lim, 2021-07-20:
@brunopinto900 Looks good, but please make a pull request, so that we can discuss the details