Jaeyoung-Lim/mavros_controllers #155: Pre Flight Checks fail on sitl_multi_vehicle_track_circle.launch

shonigmann, 2020-10-08:
I am having issues getting the multi vehicle geometric control example (using `sitl_multi_vehicle_track_circle.launch`) working on my system. The single vehicle case works without any issue, but the drones in the multi-uav case won't take off. I don't see any errors or warnings on spawn or on running the launch file. But once the drones are ready to be armed, they continuously fail their preflight checks:
```
[ INFO] [1602174051.749217238, 4.180000000]: Got pose! Drone Ready to be armed.
[ INFO] [1602174051.749218338, 4.180000000]: Got pose! Drone Ready to be armed.
[ INFO] [1602174053.574418694, 6.000000000]: Offboard enabled
[ INFO] [1602174053.575532814, 6.004000000]: Offboard enabled
INFO  [ecl/EKF] 5880000: reset position to GPS
INFO  [ecl/EKF] 5880000: reset velocity to GPS
INFO  [ecl/EKF] 5880000: starting GPS fusion
INFO  [ecl/EKF] 5876000: reset position to GPS
INFO  [ecl/EKF] 5876000: reset velocity to GPS
INFO  [ecl/EKF] 5876000: starting GPS fusion
INFO  [tone_alarm] home set
ERROR [tone_alarm] notify negative
INFO  [tone_alarm] home set
ERROR [tone_alarm] notify negative
WARN  [PreFlightCheck] Preflight Fail: Compass #1 197644 uncalibrated
WARN  [PreFlightCheck] Preflight Fail: Accel #0 uncalibrated
WARN  [PreFlightCheck] Preflight Fail: Accel #1 uncalibrated
WARN  [PreFlightCheck] Preflight Fail: Accel #2 uncalibrated
WARN  [PreFlightCheck] Preflight Fail: Gyro #0 uncalibrated
WARN  [PreFlightCheck] Preflight Fail: Gyro #1 uncalibrated
WARN  [PreFlightCheck] Preflight Fail: Gyro #2 uncalibrated
WARN  [PreFlightCheck] system power unavailable
WARN  [PreFlightCheck] Preflight Fail: Compass #1 197644 uncalibrated
WARN  [PreFlightCheck] Preflight Fail: Accel #0 uncalibrated
WARN  [PreFlightCheck] Preflight Fail: Accel #1 uncalibrated
WARN  [PreFlightCheck] Preflight Fail: Accel #2 uncalibrated
WARN  [PreFlightCheck] Preflight Fail: Gyro #0 uncalibrated
ERROR [tone_alarm] notify negative
WARN  [PreFlightCheck] Preflight Fail: Gyro #1 uncalibrated
WARN  [PreFlightCheck] Preflight Fail: Gyro #2 uncalibrated
WARN  [PreFlightCheck] system power unavailable
ERROR [tone_alarm] notify negative
ERROR [tone_alarm] notify negative
ERROR [tone_alarm] notify negative
```

I just reinstalled and rebuilt the mavros_controllers libraries and PX4 yesterday, but no luck. I also tried the two solutions suggested [here](https://github.com/PX4/Firmware/issues/12469) (running `make clean` and rebuilding, and removing `~/.ros/eeprom/parameters`.

Thanks for the great library and thanks in advance for any hints you can offer!

shonigmann, 2020-10-08:
ahhh nevermind! 

I had missed the fact that additional `parameters` files had been generated for each iris instance (under `~/.ros/sitl_iris_0/eeprom' and '~/.ros/sitl_iris_1/eeprom' respectively in case anyone runs into the same issue). Deleting those fixed everything for me!

closed 2020-10-08 by shonigmann

GitHub-SSS, 2023-04-02:
Hello, I have encountered the same problem as you. When I control my iris in QGC, it always appears that the aircraft is not unlocked.

`[ WARN] [1680414194.469823949, 129.996000000]: CMD: Command 400 -- wait ack timeout
WARN  [PreFlightCheck] Preflight Fail: Compass 1 uncalibrated
WARN  [PreFlightCheck] Preflight Fail: Accel 0 uncalibrated
WARN  [PreFlightCheck] Preflight Fail: Accel 1 uncalibrated
WARN  [PreFlightCheck] Preflight Fail: Accel 2 uncalibrated
WARN  [PreFlightCheck] Preflight Fail: Gyro 0 uncalibrated
WARN  [PreFlightCheck] Preflight Fail: Gyro 1 uncalibrated
WARN  [PreFlightCheck] Preflight Fail: Gyro 2 uncalibrated
INFO  [tone_alarm] notify negative
INFO  [tone_alarm] battery warning (fast)
WARN  [PreFlightCheck] Preflight Fail: Compass 1 uncalibrated
WARN  [PreFlightCheck] Preflight Fail: Accel 0 uncalibrated
WARN  [PreFlightCheck] Preflight Fail: Accel 1 uncalibrated
WARN  [PreFlightCheck] Preflight Fail: Accel 2 uncalibrated
WARN  [PreFlightCheck] Preflight Fail: Gyro 0 uncalibrated
WARN  [PreFlightCheck] Preflight Fail: Gyro 1 uncalibrated
WARN  [PreFlightCheck] Preflight Fail: Gyro 2 uncalibrated
INFO  [tone_alarm] notify negative
WARN  [PreFlightCheck] Preflight Fail: Compass 1 uncalibrated
WARN  [PreFlightCheck] Preflight Fail: Accel 0 uncalibrated
WARN  [PreFlightCheck] Preflight Fail: Accel 1 uncalibrated
WARN  [PreFlightCheck] Preflight Fail: Accel 2 uncalibrated
WARN  [PreFlightCheck] Preflight Fail: Gyro 0 uncalibrated
WARN  [PreFlightCheck] Preflight Fail: Gyro 1 uncalibrated
WARN  [PreFlightCheck] Preflight Fail: Gyro 2 uncalibrated
INFO  [tone_alarm] notify negative
INFO  [commander] Failsafe mode deactivated
INFO  [commander] Failsafe mode deactivated
INFO  [commander] Failsafe mode deactivated
INFO  [commander] Failsafe mode deactivated
INFO  [tone_alarm] battery warning (fast)
INFO  [tone_alarm] battery warning (fast)
INFO  [commander] Failsafe mode activated
INFO  [navigator] RTL HOME activated
INFO  [tone_alarm] battery warning (fast)
INFO  [commander] Failsafe mode activated
INFO  [navigator] RTL HOME activated
INFO  [tone_alarm] battery warning (fast)
INFO  [commander] Failsafe mode activated
INFO  [tone_alarm] battery warning (fast)
INFO  [navigator] RTL HOME activated
INFO  [commander] Failsafe mode activated
INFO  [navigator] RTL HOME activated
INFO  [tone_alarm] battery warning (fast)
`

I don't know if you have a solution, looking forward to your reply.