Jaeyoung-Lim/mavros_controllers #160: Apply to real drone + thrust_scaling parameter is set to zero

coldhenry, 2020-11-09:
Hi, thanks for the great project.

I've successfully run the sitl launch file and I would like to apply to a real drone (QAV 250).
I tried to run the 2 launch files with the only modification: `<arg name="fcu_url" default="/dev/ttyUSB0:57600"/>`
```
roslaunch geometric_controller sitl_trajectory_track_circle.launch
roslaunch geometric_controller mav_trajectory_track_circle.launch
``` 
Both have some warns about thrust_scaling parameter like following:
**In sitl_trajectory_track_circle.launch**
[image]
**In mav_trajectory_track_circle.launch**
[image]

And I've looked through the issue #83 so I tried to reinstall mavros from source but no luck.
So there are 2 questions I would like to ask:
1. What steps do I miss to apply the controller to a real drone other than modifying the fcu_url?
2. Is there any idea about the warnings?

**Version and platform**
ROS: Melodic
Ubuntu: 18.04
PX4 Version: 1.11

**Diagnostics**
```
helios300:~/catkin_ws/src/mavros$ rostopic echo -n1 /diagnostics 
header: 
  seq: 17
  stamp: 
    secs: 1604882755
    nsecs: 485414936
  frame_id: ''
status: 
  - 
    level: 0
    name: "mavros: FCU connection"
    message: "connected"
    hardware_id: "/dev/ttyACM0:57600"
    values: 
      - 
        key: "Received packets:"
        value: "6762"
      - 
        key: "Dropped packets:"
        value: "0"
      - 
        key: "Buffer overruns:"
        value: "0"
      - 
        key: "Parse errors:"
        value: "0"
      - 
        key: "Rx sequence number:"
        value: "12"
      - 
        key: "Tx sequence number:"
        value: "0"
      - 
        key: "Rx total bytes:"
        value: "382728"
      - 
        key: "Tx total bytes:"
        value: "84230"
      - 
        key: "Rx speed:"
        value: "22544.000000"
      - 
        key: "Tx speed:"
        value: "5719.000000"
  - 
    level: 2
    name: "mavros: GPS"
    message: "No satellites"
    hardware_id: "/dev/ttyACM0:57600"
    values: 
      - 
        key: "Satellites visible"
        value: "0"
      - 
        key: "Fix type"
        value: "0"
      - 
        key: "EPH (m)"
        value: "Unknown"
      - 
        key: "EPV (m)"
        value: "Unknown"
  - 
    level: 0
    name: "mavros: Heartbeat"
    message: "Normal"
    hardware_id: "/dev/ttyACM0:57600"
    values: 
      - 
        key: "Heartbeats since startup"
        value: "16"
      - 
        key: "Frequency (Hz)"
        value: "1.000002"
      - 
        key: "Vehicle type"
        value: "Quadrotor"
      - 
        key: "Autopilot type"
        value: "PX4 Autopilot"
      - 
        key: "Mode"
        value: "MANUAL"
      - 
        key: "System status"
        value: "Standby"
  - 
    level: 2
    name: "mavros: System"
    message: "Sensor health"
    hardware_id: "/dev/ttyACM0:57600"
    values: 
      - 
        key: "Sensor present"
        value: "0x122E002F"
      - 
        key: "Sensor enabled"
        value: "0x1221002F"
      - 
        key: "Sensor health"
        value: "0x002E000F"
      - 
        key: "3D gyro"
        value: "Ok"
      - 
        key: "3D accelerometer"
        value: "Ok"
      - 
        key: "3D magnetometer"
        value: "Ok"
      - 
        key: "absolute pressure"
        value: "Ok"
      - 
        key: "GPS"
        value: "Fail"
      - 
        key: "rc receiver"
        value: "Fail"
      - 
        key: "AHRS subsystem health"
        value: "Ok"
      - 
        key: "Battery"
        value: "Fail"
      - 
        key: "CPU Load (%)"
        value: "48.4"
      - 
        key: "Drop rate (%)"
        value: "0.0"
      - 
        key: "Errors comm"
        value: "0"
      - 
        key: "Errors count #1"
        value: "0"
      - 
        key: "Errors count #2"
        value: "0"
      - 
        key: "Errors count #3"
        value: "0"
      - 
        key: "Errors count #4"
        value: "0"
  - 
    level: 0
    name: "mavros: Battery"
    message: "Normal"
    hardware_id: "/dev/ttyACM0:57600"
    values: 
      - 
        key: "Voltage"
        value: "65.54"
      - 
        key: "Current"
        value: "-0.0"
      - 
        key: "Remaining"
        value: "-1.0"
  - 
    level: 0
    name: "mavros: Time Sync"
    message: "Normal"
    hardware_id: "/dev/ttyACM0:57600"
    values: 
      - 
        key: "Timesyncs since startup"
        value: "160"
      - 
        key: "Frequency (Hz)"
        value: "10.000023"
      - 
        key: "Last RTT (ms)"
        value: "0.635409"
      - 
        key: "Mean RTT (ms)"
        value: "0.863344"
      - 
        key: "Last remote time (s)"
        value: "689.127082000"
      - 
        key: "Estimated time offset (s)"
        value: "1604882066.343814850"
---

```

Any suggestion is appreciated!

Jaeyoung-Lim, 2020-11-09:
@coldhenry First of all, if your baudrate is set to 57600, it is way too low to have any kind of offboard control working. How is your computer running ROS connected to the fmu? The recommended rate is 921600, therefore usually this means you need a serial connection with a onboard computer

coldhenry, 2020-11-13:
Thanks for the prompt response. I launched the geometric controller in my computer and tried to connect to the fmu with SiK radios. If I understand correctly, do you mean that it is impossible to use telemetry radio because the baud rate is too low? Thank you for the clarification!

Jaeyoung-Lim, 2020-11-13:
Yes, it is impossible - You need at least a few tens of hz to control the vehicle, which the telemetry radio is unable to send

closed 2020-12-01 by coldhenry