Jaeyoung-Lim/mavros_controllers #161: process has died [pid 9907, exit code -6]

coldhenry, 2020-11-26:
Hello Jaeyoung,

I am working on the installation of this project on a companion computer as you advised in #160 and there is another issue coming out during roslaunch.

[image]

**Version and platform**
Companion computer: Odroid XU4
ROS: Melodic
Ubuntu: 18.04 Bionic
PX4 Version: 1.11

Any thought is appreciated!

Jaeyoung-Lim, 2020-12-24:
Sorry for the late response.

Are you sure you are running the latest `master`? I just ran the setup, and it is working as expected

Jaeyoung-Lim, 2021-01-22:
@coldhenry Did you manage to solve your problem?

Jaeyoung-Lim, 2021-02-02:
Closing since original author didn't respond.

Please reopen if you feel like this is still an issue @coldhenry

closed 2021-02-02 by Jaeyoung-Lim