Jaeyoung-Lim/mavros_controllers #165: Kvel_ initalitation error?

pluto-project-it, 2021-02-04:
In geometric_controller.cpp line 489:
Kvel_ << -Kvel_x_, -Kvel_z_, -Kvel_z_;
Shouldn't it be:
Kvel_ << -Kvel_x_, -Kvel_y_, -Kvel_z_;
?
Thank you for your work.
Giorgio Vaccarino

Jaeyoung-Lim, 2021-02-04:
@pluto-project-it I think that looks like an error. Would you mind making a PR for the fix?

pluto-project-it, 2021-02-04:
We are working to manage multiple MultiDOFJointTrajectoryPoint and we have heavily modified the code and is not stable yet.
Said that if you want I can do a PR with everything as memo.

Jaeyoung-Lim, 2021-02-04:
@pluto-project-it Not sure if I follow, the ask was not to pull in all your changes, but only the fix you mentioned in the original issue

pluto-project-it, 2021-02-05:
Did in https://github.com/Jaeyoung-Lim/mavros_controllers/pull/166#issue-568190625

closed 2021-02-05 by Jaeyoung-Lim