Jaeyoung-Lim/mavros_controllers #169: Edit Trajectory

dedyebhi, 2021-02-14:
how do you edit the trajectory circle program, sir? because my final project in college I used a square trajectory. Thank you sir

Jaeyoung-Lim, 2021-02-14:
@dedyebhi The trajectory shapes are defined in https://github.com/Jaeyoung-Lim/mavros_controllers/blob/393242d0a42904383b5bb93e09b5dc620eab1d43/trajectory_publisher/src/shapetrajectory.cpp#L70

You can add a square trajectory as another "shape"

dedyebhi, 2021-02-14:
> @dedyebhi The trajectory shapes are defined in
> 
> https://github.com/Jaeyoung-Lim/mavros_controllers/blob/393242d0a42904383b5bb93e09b5dc620eab1d43/trajectory_publisher/src/shapetrajectory.cpp#L70
> 
> You can add a square trajectory as another "shape"

Can you help me create a program to add a squere trajectory?

Jaeyoung-Lim, 2021-02-14:
@dedyebhi Is there more information that you need? where are you stuck?

dedyebhi, 2021-02-14:
> @dedyebhi Apakah ada lebih banyak informasi yang Anda butuhkan? dimana kamu terjebak

dedyebhi, 2021-02-14:
> @dedyebhi Is there more information that you need? where are you stuck?

I don't understand how to add a square trajectory

closed 2021-02-14 by dedyebhi

Jaeyoung-Lim, 2021-02-14:
@dedyebhi What part do you not understand? You just need to add a case for a square in

https://github.com/Jaeyoung-Lim/mavros_controllers/blob/393242d0a42904383b5bb93e09b5dc620eab1d43/trajectory_publisher/src/shapetrajectory.cpp#L70-L99

dedyebhi, 2021-02-14:
> @dedyebhi What part do you not understand? You just need to add a case for a square in
> 
> https://github.com/Jaeyoung-Lim/mavros_controllers/blob/393242d0a42904383b5bb93e09b5dc620eab1d43/trajectory_publisher/src/shapetrajectory.cpp#L70-L99

can you reason me for example?

Jaeyoung-Lim, 2021-02-14:
@dedyebhi You can use the circle trajectory as an example. I specifically linked the code as an example

dedyebhi, 2021-02-14:
Eigen::Vector3d shapetrajectory::getPosition(double time) {
  Eigen::Vector3d position;
  double theta;

  switch (type_) {
    case TRAJ_ZERO:

      position << 0.0, 0.0, 0.0;
      break;

    case TRAJ_CIRCLE:

      theta = traj_omega_ * time;
      position = std::cos(theta) * traj_radial_ + std::sin(theta) * traj_axis_.cross(traj_radial_) +
                 (1 - std::cos(theta)) * traj_axis_.dot(traj_radial_) * traj_axis_ + traj_origin_;
      break;

    case TRAJ_LAMNISCATE:  // Lemniscate of Genero

      theta = traj_omega_ * time;
      position = std::cos(theta) * traj_radial_ + std::sin(theta) * std::cos(theta) * traj_axis_.cross(traj_radial_) +
                 (1 - std::cos(theta)) * traj_axis_.dot(traj_radial_) * traj_axis_ + traj_origin_;
      break;

    case TRAJ_SQUERE:

      theta = traj_omega_ * time;
      position = std::cos(theta) * traj_radial_ + std::sin(theta) * std::cos(theta) * traj_axis_.cross(traj_radial_) +
                 (1 - std::cos(theta)) * traj_axis_.dot(traj_radial_) * traj_axis_ + traj_origin_;
      break;

    case TRAJ_STATIONARY:  // Lemniscate of Genero

      position = traj_origin_;
      break;
  }
  return position;
}

dedyebhi, 2021-02-14:
> Eigen::Vector3d shapetrajectory::getPosition(double time) {
> Eigen::Vector3d position;
> double theta;
>
> switch (type_) {
> case TRAJ_ZERO:
>
> ```
>   position << 0.0, 0.0, 0.0;
>   break;
>
> case TRAJ_CIRCLE: [...]

Is it like this ?

Jaeyoung-Lim, 2021-02-14:
@dedyebhi Yes, but you would need to write the square part

dedyebhi, 2021-02-14:
> @dedyebhi Yes, but you would need to write the square part

Can you help me, what is squere part like?

Jaeyoung-Lim, 2021-02-14:
@dedyebhi I have pointed you to the relevant places where you can modify this package. Drawing the square trajectory seems to be part of what you are doing, and not related to this package. Therefore, unfortunately I think that is the part that you need to figure out yourself.

dedyebhi, 2021-02-14:
> @dedyebhi Saya telah mengarahkan Anda ke tempat-tempat yang relevan di mana Anda dapat memodifikasi paket ini. Menggambar lintasan persegi tampaknya menjadi bagian dari apa yang Anda lakukan, dan tidak terkait dengan paket ini. Oleh karena itu, sayangnya saya pikir itu adalah bagian yang perlu Anda pikirkan sendiri.

please, i need your help

dedyebhi, 2021-02-14:
> @dedyebhi I have pointed you to the relevant places where you can modify this package. Drawing the square trajectory seems to be part of what you are doing, and not related to this package. Therefore, unfortunately I think that is the part that you need to figure out yourself.

please, i need your help

Jaeyoung-Lim, 2021-02-14:
@dedyebhi I have already helped you!

I cannot implement your project for you. Since the feature you are requesting is not something relevant to this package, you would need to figure how to implement a square yourself.

Please feel free to ask if you have any questions regarding this package.

dedyebhi, 2021-02-14:
> @dedyebhi I have already helped you!
> 
> I cannot implement your project for you. Since the feature you are requesting is not something relevant to this package, you would need to figure how to implement a square yourself.
> 
> Please feel free to ask if you have any questions regarding this package.

can you send me a reference regarding the square trajectory quadcopter?