Jaeyoung-Lim/mavros_controllers #173: PX4 version needed for geometric_controller

sriram-b, 2021-02-28:
Hi,

   In #156 you noted that the iris motor thrust changed in a recent release of PX4 firmware, and that the geometric_controller parameters needed re-tuning as a result. So I wanted to know, which version of the PX4 firmware are the parameters in geometric_controller tuned for? Or am I misunderstanding what you meant?

Thank you in advance.

Jaeyoung-Lim, 2021-03-01:
@sriram-b with latest mastee, it works without any additional tuning. Are you asking becuase it is not working for you?

sriram-b, 2021-03-01:
@Jaeyoung-Lim 

Thanks for the quick response. I'm in a time crunch currently so I would really appreciate it if you could do that again. 

I was asking because I am currently doing a project where the drone does not seem to be tracking the given trajectory very closely. To test my setup, I tried cloning the repository and running, without any modification, the [lamniscate example](https://utexas.box.com/s/fxgii9juoa5wn4yq2s1ooekf0xne8w97). The tracking seems to be very poor, and not only that, the drone will eventually just crash after only a couple of minutes. The only error from the terminal was [this](https://utexas.box.com/s/n7l4001mm8exc70thdz2r7u0m3kswro2).  

On the other hand, the [circle example](https://utexas.box.com/s/k1310y2951xgmkxamqr7n4qbsvcy99fc) ran much better, with no errors that I could tell.

Is there anything wrong with my setup? Since the geometric controller is tuned to the iris drone with the latest stable PX4 release, I was wondering what I can do to make the tracking better.

Jaeyoung-Lim, 2021-03-01:
@sriram-b I am not sure what you are asking me to do "again" - it is already tuned

The lamniscate trajectory is in general a much more challenging trajectory to track. You can slow the trajectory speed to make it less challenging

Probably adding the acceleration reference will improve the tracking performance

sriram-b, 2021-03-01:
@Jaeyoung-Lim 

I am unfamiliar with the acceleration reference, how can I add that?

Jaeyoung-Lim, 2021-03-01:
@sriram-b https://github.com/Jaeyoung-Lim/mavros_controllers/blob/07d0203290c0bdd97a21389db2bdf6d3b56077ef/trajectory_publisher/src/shapetrajectory.cpp#L143

Have you tried lowering the speed? I in purpose made the lamniscate trajectory hit the limit of the controller.

sriram-b, 2021-03-01:
@Jaeyoung-Lim 

Lowering the speed does help, it works well after that. I was under the impression that it should've worked out of the box, so that's a slight relief.

I'll look into adding the acceleration reference then and make another issue if I need to ask anything else, thank you!

closed 2021-03-01 by sriram-b