Jaeyoung-Lim/mavros_controllers #174: Compatibility with Ardupilot Copter Firmware

Kavin-Kailash, 2021-03-10:
Thanks for the great work. 

From the files, it is apparent that this package is intended for use with PX4 mainly. I was wondering about the compatibility with the Ardupilot Flight Stack.

Jaeyoung-Lim, 2021-03-13:
@Kavin-Kailash Since it is using mavlink it should be overall compatible. However, there might be some subtle differences on how offboard control works and this has never been tested.

Kavin-Kailash, 2021-03-13:
@Jaeyoung-Lim Thanks for the quick reply. From the documentation I could infer that the final setpoints sent to PX4 are mavros_msgs/AttitudeTarget and Pose Stamped , is there any way to configure geometric_controller so that in only provides a specific setpoint?

Jaeyoung-Lim, 2021-03-13:
@Kavin-Kailash Yes you can. Depending on the setpoint you want to use, you can use either the trajectory_publisher or the geometric_controller

closed 2021-03-22 by Jaeyoung-Lim