Jaeyoung-Lim/mavros_controllers #175: A bug about the reference acceleration in geometric controller

shupx, 2021-03-24:
Thanks for your great work!
It seems like there is a bug in geometric_controller.cpp
the relevant code is below:

https://github.com/Jaeyoung-Lim/mavros_controllers/blob/2f0feb62ee638952be58798cc4d3a45811592adb/geometric_controller/src/geometric_controller.cpp#L156-L159
the line 157 is wrong, and it should be:
`targetAcc_ = toEigen(msg.acceleration);`

here the geometric controller receives reference setpoint from topic "reference/flatsetpoint". The msg.type_mask is defined as below:
`uint8 IGNORE_SNAP=1`
`uint8 IGNORE_SNAP_JERK=2`
`uint8 IGNORE_SNAP_JERK_ACC=4`
so when the msg.type_mask == 2, the acceleration should be received rather than set to be zero.
this bug has caused that the geometric controller does not receive the reference acceleration when tracking a circle. so there is always a position error no matter how i adjust the gain. the following figure illustrate this:
[image]

After I fix this bug, there is no position error anymore. The tracking performance is shown below:
[image]

Jaeyoung-Lim, 2021-03-24:
@shupx Thanks! Would it be possible for you add a pull request with the fix?

shupx, 2021-03-24:
> @shupx Thanks! Would it be possible for you add a pull request with the fix?

 I have made a pull request

PR #176 (2021-03-24): Update geometric_controller.cpp
Fixes #175
closed 2021-03-24 by Jaeyoung-Lim