Jaeyoung-Lim/mavros_controllers #179: Controller cannot track reference trajectory and travels upwards infinitely

rptamin, 2021-05-23:
I am attempting to integrate the geometric controller into AirSim and I am using PX4 version v1.11.3. I am running into an issue where the controller cannot hold altitude. I am currently using the circle reference trajectory given in this repo. The overall trajectory looks like a poorly tracking helix. I reduced the norm thrust constant along with offset, but it still cannot keep altitude. How would I go about tuning this?

Jaeyoung-Lim, 2021-05-23:
@Jaeyoung-Lim Please provide a log and how you are running this package.

Most likely it is the thrust scaling not being tuned ( you can check the logs to verify this).

However, the information you have provided is not enough to make any colclusions

rptamin, 2021-05-23:
Here is an upload of the logs:
https://review.px4.io/plot_app?log=5229a0ae-3451-4068-8fa0-20011323aa4d

Jaeyoung-Lim, 2021-05-23:
@rptamin and...how are you running the package?

rptamin, 2021-05-23:
I have launch file which is shown here: `
  <include file="$(find airsim_ros_pkgs)/launch/airsim_node.launch"/>
    
  
  <include file="$(find exploration_manager)/launch/rviz.launch" />

	<arg name="gcs_url" default="" />
	<arg name="tgt_system" default="1" />
	<arg name="tgt_component" default="1" />
	<arg name="log_output" default="screen" />
	<arg name="fcu_protocol" default="v2.0" />
	<arg name="respawn_mavros" default="false" /> 

  
  <include file="$(find px4_fast_planner)/launch/geometric_controller.launch">
        <arg name="gazebo_simulation" value="$(arg use_gazebo)" />
        <arg name="max_acc" value="$(arg max_acc)" />
        <param name="Kp_x" value="20" />
        <param name="Kp_y" value="20" />
        <param name="Kp_z" value="5" />
        <param name="Kv_x" value="2" />
        <param name="Kv_y" value="2" />
        <param name="Kv_z" value=".5" />
        <param name="ctrl_mode" value="2" />
        <param name="attctrl_constant" value="4" />
        <param name="normalizedthrust_constant" value=".01" />
        <param name="normalizedthrust_offset" value =".01" />
        
        <arg name="init_pos_x" value="0.0" />
        <arg name="init_pos_y" value="0.0" />
        <arg name="init_pos_z" value="2.5" />
    </include>
    <node pkg="trajectory_publisher" type="trajectory_publisher" name="trajectory_publisher" output="screen">
        <param name="trajectory_type" value="1" />
        <param name="shape_omega" value="1.2" />
        <param name="initpos_z" value="2.0" />
        <param name="reference_type" value="2" />
  </node>  
	<include file="$(find mavros)/launch/node.launch">
		<arg name="pluginlists_yaml" value="$(find px4_fast_planner)/config/px4_pluginlists.yaml" />
		<arg name="config_yaml" value="$(find px4_fast_planner)/config/px4_config.yaml" />

		<arg name="fcu_url" value="$(arg fcu_url)" />
		<arg name="gcs_url" value="$(arg gcs_url)" />
		<arg name="tgt_system" value="$(arg tgt_system)" />
		<arg name="tgt_component" value="$(arg tgt_component)" />
		<arg name="log_output" value="$(arg log_output)" />
		<arg name="fcu_protocol" value="$(arg fcu_protocol)" />
		<arg name="respawn_mavros" default="$(arg respawn_mavros)" />
	</include>`

rptamin, 2021-05-23:
@Jaeyoung-Lim I should note that I have also messed around with the attctrl_constant. It was ~ .1 in previous tests but still had the same issue. I also tried to use ctrl_mode = 1. It looks like there is sustained acceleration in the z direction when looking at the raw acceleration chart.

Jaeyoung-Lim, 2021-05-23:
@rptamin If you look at the log, the actuator outputs are all saturated. Have you tried the default launchfile?

rptamin, 2021-05-23:
Thank you! I reset every parameter how it is in repo, and it worked! My max acceleration was set to .5. That may have been the error.

closed 2021-05-23 by rptamin