Jaeyoung-Lim/mavros_controllers #182: What are norm_thrust_const_ and norm_thrust_offset_ for?

mzahana, 2021-06-30:
Hi @Jaeyoung-Lim 
Could you please provide some explanation on what [norm_thrust_offset_](https://github.com/Jaeyoung-Lim/mavros_controllers/blob/master/geometric_controller/src/geometric_controller.cpp#L94) and [norm_thrust_const_](https://github.com/Jaeyoung-Lim/mavros_controllers/blob/master/geometric_controller/src/geometric_controller.cpp#L93) are used for?
Is one of them related to the vehicle mass?
How can they be tuned?
Thanks

mzahana, 2021-06-30:
Also, what is the effect of [attctrl_constant](https://github.com/Jaeyoung-Lim/mavros_controllers/blob/master/geometric_controller/src/geometric_controller.cpp#L92)?

Jaeyoung-Lim, 2021-06-30:
> Could you please provide some explanation on what norm_thrust_offset_ and norm_thrust_const_ are used for?

It is the linear constants that maps the thrust values(N) into normalized thrust(0 ~ 1)

> Also, what is the effect of attctrl_constant?

You can check `tau` is (which is a time cosntant for attitude control in Brescianini, Dario, Markus Hehn, and Raffaello D'Andrea. Nonlinear quadrocopter attitude control: Technical report. ETH Zurich, 2013.

mzahana, 2021-07-01:
@Jaeyoung-Lim Is there a recommended way of tuning such parameters for different multi-rotors?

Jaeyoung-Lim, 2021-07-01:
@mzahana You are just mapping your thurst (N) to normalized thrust values. you can do this by measuring your thrust, or simply hand tuning it

closed 2021-07-11 by Jaeyoung-Lim