Jaeyoung-Lim/mavros_controllers #184: AirSim

brunopinto900, 2021-07-08:
Hello,

Thank you for the amazing code. I intend to use this as my controller for autonomous drone racing.

Can this repo work with AirSim instead of gazebo? 

Thank you.

Jaeyoung-Lim, 2021-07-08:
@brunopinto900 Thanks for the nice complement. Sure, this package directly interfaces with PX4, so the simulation framework doesn't matter

closed 2021-07-11 by Jaeyoung-Lim

Zartris, 2021-07-21:
Well this gives me some problems as I'm getting "Resource not found: gazebo_ros" so there must be some dependencies

Jaeyoung-Lim, 2021-07-21:
@Zartris 
```
apt install ros-<ros_distro>-gazebo-ros
```

Zartris, 2021-07-21:
Yes I got it working, I was just mentioning the dependency :) 
I just set the gazebo parameters in the launch file to be false and it fixed it.

Jaeyoung-Lim, 2021-07-21:
@Zartris Great, anyway this seems to be unrelated to this issue