Jaeyoung-Lim/mavros_controllers #185: From Sim Drone to Real Drone - Some doubts/issues

edwinpha, 2021-07-14:
Hi, first of all, thanks for making public such a great project.

I have some doubts related to how to tune some constants. Particularly, the norm_th and norm_offset.

In gazebo simulation, I have noticed a gap in z of around 15-20 cm less than the expected hovering point (even variating it). As mentioned in #156, the motor coefficient has changed and this can be solved with a slight variation of either norm_th or norm_offset.

Now, I have tested it with a real drone. I could see the effect of the wrong choice of norm_th and norm_offset, because I have a gap in z much more significant than 20 cm. Just changing the battery, the performance in the z-axis changed significantly (battery 80 gr lighter). 
The X-Y plane is working perfectly and the drone is really steady so I believe most of the gains are correctly tuned. (This happens with both batteries).

Different from tuning the drone in simulation, where manual tuning is an easy and fast option, in a real drone I would like to follow a more mathematical way as a starting point and then manually tune from there if needed. To avoid of course unexpected crashes.

I tried to map thrust and acceleration as @HenryHuYu did in #156, but my acceleration does not variate so strongly as his (5 - 15 ms^-2), and for this reason, my eq. y = m*x + b has a big coefficient b but a not-so-steady slope. This may be an option because my hover thrust is around 70%.

Still, there is something bothering me because I tested this relation among thrust and acceleration in the simulator. Aiming to obtain values closer to the ones in the code. And I got totally different values. Showing me that most probably I am calculating them wrong. 

If you can give me any advice about the tuning it would be great. Maybe how you obtained the params for gazebo simulation, when writing the code, and from there I can analyze the current parameters. Actually, I would like to work on a testbed for correctly modeling the motor/prop pair.

Also, I was wondering. In a real drone, considering that battery power is consuming while flying, the "new" hover thrust is going to increase with time. Is this considered in any way or is just me that has a wrong concept. Because I tried to obtain the values by manually flying it and keeping it in a hovering position for the whole duration of the battery. When watching the log files the thrust increases from 0.7 to 1 because the battery is discharging.

closed 2021-08-23 by edwinpha

Jaeyoung-Lim, 2021-08-23:
@edwinpha Sorry that I missed this issue, did you manage to figure everything out?

edwinpha, 2021-08-24:
@Jaeyoung-Lim Yes! It took me a little while because I needed to be very sure (did not want to risk the drone), however, it did work as expected.

Jaeyoung-Lim, 2021-08-24:
@edwinpha Exciting! Could you share any videos of the vehicle flying?

edwinpha, 2021-08-24:
Sure, I modify a little bit the code (landing for example), but this video is essentially your code.
Though, I am currently working to make it more precise without losing much of its aggressivity. (Among other stuff)

https://user-images.githubusercontent.com/56879821/130593342-4e84056e-04be-491d-b5b2-448215720829.MOV

Jaeyoung-Lim, 2021-08-24:
@edwinpha Thanks! Looks awesome!

edwinpha, 2021-09-14:
Hey @Jaeyoung-Lim, I am still working on some HW tests.

When the waypoint or trajectory is far enough the controller works really well. 
However, for short movements, it is too aggressive. The drone's roll and pitch angle are really steep for movements under 1.5 meters approx.
I've tried reducing max_acc, and, Kp and Kv of axis X-Y; though it is still a little bit aggressive!

Any advice comes to your head?