Jaeyoung-Lim/mavros_controllers #187: /mavros/local_position/pose not working properly

Novotendo, 2021-07-19:
Hello @Jaeyoung-Lim 

I'm currently trying to use your package using a single GPS for position estimation as first test. The problem is that, when my drone is at the ground and I read the topic /mavros/local_position/pose, the X and Y are always 0.0. The rest of the values: Z, and the quaternion are correctly displayed. Maybe do I need to change a parameter in the pixhawk? I'm using:
-Pixhawk 2.4.8
-PX4 1.11.3

Thanks!

Jaeyoung-Lim, 2021-07-19:
@Novotendo Are you sure you have a valid position estimate (e.g. GPS fix)?

Novotendo, 2021-07-19:
Yes, I've got 3D Lock, here is a capture:
[image]

Jaeyoung-Lim, 2021-07-19:
@Novotendo Okay, and have you tried arming the vehicle?

Novotendo, 2021-07-19:
Even armed, it is still in the same way. What is considered a valid position estimate? And how could I get it?

Jaeyoung-Lim, 2021-07-19:
@Novotendo Can your vehicle fly in position mode?

You can look up docs.px4.io to look at estimator related information of PX4

Novotendo, 2021-07-19:
It is set into OFFBOARD mode

Jaeyoung-Lim, 2021-07-19:
@Novotendo If you don't have a valid position estimate, you won't be able to do anything in offboard mode. Therefore, it is important that you make sure everything works on your vehicle (flying in position / mission mode) before you put the vehicle into offboard mode.

The fact that you mention you don't have position estimates just means that your EKF is not able to have a valid position estimate. Therefore you need to fix this problem first before you interact with the vehicle with mavros

Novotendo, 2021-07-19:
And which EKF params should be set for a vehicle using just GPS data?

Jaeyoung-Lim, 2021-07-19:
@Novotendo You cannot use just GPS data. Please go through the PX4 documentation: https://docs.px4.io/master/en/advanced_config/tuning_the_ecl_ekf.html since it is explained why this is the case

Novotendo, 2021-07-20:
Yes, you were right @Jaeyoung-Lim ! And one more question: do you know how to "reset" the current position of the drone via mavros to (0,0,0)?

closed 2021-12-23 by Jaeyoung-Lim